z88dk/z88dk #2397: Cross-compiling for m68k-atari-mint on linux

th-otto, 2023-09-29:
I've just cross-compiled z88dk for m68k-atari-mint (Atari running FreeMiNT).

For that, i needed to do a few small patches (attached below). One was for compiling the z80asm parser. The generated switch table seems to be too large (m68k normally has a limit of 32k for this), no matter what optimization flags are used. Therefor, an extra option ``-mlong-jump-table-offsets`` has to be added to its compilation flags.

The other was about linking xml2. When cross-compiling, xml2-config cannot be used; this is only for the host system. Since atari uses static libs, linking with just -xml2 is not enough, you also have to link to -lz and maybe others. So the patch uses pkg-config instead, which can load its configuration files via PKG_CONFIG_PATH suitable for the target.

Attached is also a small script to automate all that. It first compiles a native version so the libraries can be build, then afterwards the the binaries only for the target. Maybe someone will find that useful.

[z88dk.patch.txt](https://github.com/z88dk/z88dk/files/12769177/z88dk.patch.txt)
[z88dk-m68k-atari-mint-build.txt](https://github.com/z88dk/z88dk/files/12769186/z88dk-m68k-atari-mint-build.txt)

suborb, 2023-09-29:
This is really cool, thank you. It's a long time since I built for m68k!

We build nightly snapshots (cross-compiled) for a number of targets so if you have a script that builds the toolchain and required ST libraries I can add it in.

th-otto, 2023-09-30:
Do you mean the compiler toolchain itself? Scripts, prebuild binaries, and also any needed libraries for linux are available at https://tho-otto.de/crossmint.php. There is also a ppa archive available for the compiler and some of the libs at http://vincent.riviere.free.fr/soft/m68k-atari-mint/ubuntu.php, however that is gcc version 4.6.4 only, dunno whether that is recent enough. I used gcc-7.5.0 in my tests.

The archives on my site are just tar archives, not packages that you could simply install using apt.

You can also look at eg. https://github.com/freemint/cflib/blob/master/.github/workflows/build.yml (and the scripts in the ~/.scripts directory) where those toolchains are used to build a library.

suborb, 2023-09-30:
I used your tarballs for 7.5.0 and unfortunately no joy:

```
make
make -C src/appmake PREFIX=`pwd` install
make[1]: Entering directory '/home/build/z88dk/mint/z88dk/src/appmake'
ccache m68k-atari-mintelf-gcc -g -O2 -fomit-frame-pointer -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o mz2500.o mz2500.c
In file included from mz2500.c:8:0:
appmake.h:24:0: warning: "__NORETURN" redefined
 #define __NORETURN __attribute((noreturn))

In file included from /usr/m68k-atari-mintelf/sys-root/usr/include/features.h:24:0,
                 from /usr/m68k-atari-mintelf/sys-root/usr/include/stdio.h:29,
                 from appmake.h:11,
                 from mz2500.c:8:
/usr/m68k-atari-mintelf/sys-root/usr/include/compiler.h:62:0: note: this is the location of the previous definition
 #define __NORETURN __attribute__ ((noreturn))

mz2500.c: In function 'write_sector':
mz2500.c:45:6: internal compiler error: Segmentation fault
 void write_sector(disc_handle *h, int track, int sector, int head)
      ^~~~~~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
make[1]: *** [<builtin>: mz2500.o] Error 1
make[1]: Leaving directory '/home/build/z88dk/mint/z88dk/src/appmake'
make: *** [Makefile:130: bin/z88dk-appmake] Error 2
```

So it looks like cc1 is crashing for me. I'll build my own version and see how things get on.

th-otto, 2023-09-30:
Hm, yes, they are currently a bit hackish. Problem is that my system (SuSE tumbleweed) is too new for most ubuntu/debian releases. It uses glibc 2.38, and when i compile programs normally there, they don't work on systems with older glibc. I tried too work around that, but maybe something is still going wrong.

BTW, you should use the normal (m68k-atari-mint) toolchains. The *-mintelf are a bit experimental (they use elf object format, but the atari executable format is treated as a.out because it is very similar). Maybe that will change in the near future, and the mintelf toolchains will produce (almost) elf executables.

Thanks you very much for taking your time to look into this.

th-otto, 2023-10-04:
Finally managed to get a workflow running. You will find it at https://github.com/th-otto/z88dk/blob/master/.github/workflows/build-atari-on-ubuntu.yml

It needs a few patches (mostly from the z88dk.patch.txt above, and an additional one for the pthread library that is missing):
- https://github.com/th-otto/z88dk/commit/c116d1a82c5c19f565b1edf7e5f726d1a98fc403
- https://github.com/th-otto/z88dk/commit/9227e088714fa290642252540967ddbd4113d0a2
- https://github.com/th-otto/z88dk/commit/91f2d08044f47834fb138fc707167d197275aeac
- https://github.com/th-otto/z88dk/commit/ffd178589b18c1efa85015d20bc516ac2fcf4925

(if you want i can also put them on a different branch and create a PR from it).

The workflow uses ubuntu 20.04 (not ubuntu-latest, which is imho currently 22.04), because that was the system the cross-compiler was build on. Maybe they work in 22.04 too, didn't try yet. The archives (and some system libraries), are currently downloaded from my server. gcc is version 9.5.0 (gcc-7 does not work because it does not support std::filesystem, and gcc-13 did not work because the host gcc of ubuntu (also gcc-9) was not able to compile it.

The missing pthread library means that z88dk-gdb can't be built. There is one available which is good enough to compile z88dk-gdb, but it is based on pth, that means it is a cooperative implementation, and threads are not preempted by the kernel. The test is currently just based on $(CC), maybe there are better ways to check for -lpthread.
z88dk-ticks does not seem to need the pthread library, although  $(GDBLDFLAGS) was also added there.

suborb, 2023-10-04:
Thank you so much, that's totally amazing.

By complete coincidence, the nightly build machine is 20.04 and I've downloaded your tarballs and this time the compiler works.

I'll incorporate the changes - most of them will go in with no side-effects just the xml2-config -> pkg-config change  that I need to make sure works for other toolchains we use.

suborb, 2023-10-05:
I've merged in and poked a nightly build to produce a kit, so it should be available from: http://nightly.z88dk.org/z88dk-atari-mint-latest.zip

A few years ago we had this problem: https://github.com/z88dk/z88dk/issues/1451 and I can't remember how @pauloscustodio fixed it. However if is just reading the .lib files into a buffer and then parsing on every symbol search then that's not going to perform too well on a low Mhz m68k. On the other hand, keeping a parsed version of every symbol may use up too much memory!

pauloscustodio, 2023-10-05:
> A few years ago we had this problem: https://github.com/z88dk/z88dk/issues/1451 and I can't remember how @pauloscustodio fixed it.

The fix was allocating memory with malloc() for all the libraries and object files being linked, fread() them into memory and do all the parsing in memory.

I did not use ```mmap()``` to keep the code portable.

th-otto, 2023-10-05:
Many thanks! 

I guess the nightly builds are run somewhere else? Atleast i could not find any artifacts in the workflow builds.

The problem with the libs seems to be what the user also encountered: https://www.atari-forum.com/viewtopic.php?p=451872#p451872

So yes of course, you can't expect it to build in the same time as on a modern system. But linking taking 2min for a source of 20k seems to be quite alot, and his Milan is a "high end" machine for m68k.
One thing he suggests is to strip local (non-exported) symbols from the libraries. But of course that will make debugging harder.

suborb, 2023-10-05:
> The fix was allocating memory with malloc() for all the libraries and object files being linked, fread() them into memory and do all the parsing in memory.

Is the parsing done once on load or everytime a symbol is searched for?

suborb, 2023-10-05:
> I guess the nightly builds are run somewhere else? Atleast i could not find any artifacts in the workflow builds.

Yeah, the builds are run in a container that is hosted on my desk :) Not ideal, but there you go.

> So yes of course, you can't expect it to build in the same time as on a modern system. But linking taking 2min for a source of 20k seems to be quite alot

I agree - way back when, I used a 25Mhz 68040 for z88dk

pauloscustodio, 2023-10-05:
> Is the parsing done once on load or everytime a symbol is searched for?

A hash table is created for each object file containing all the defined symbols in that object file. The object file is then parsed only if there is a symbol match.

pauloscustodio, 2023-10-05:
st_add_string searches/creates the entry in the symbols hash table.

th-otto, 2023-10-05:
Hm, but 528481 calls to that function? Could it be that the linker searches *all* libraries in the directory?

pauloscustodio, 2023-10-05:
No, it searches only the ones passed in the command line. It does multiple passes, after each object file is linked in, it searches again the list, so that the modules are included in the order passed in the command line.

suborb, 2023-10-08:
I'm going to close this issue, lets open a new one if we can get to the bottom of the linking speed issue.

Unfortunately, for some reason the MacAranym is a bizarrely offset keyboard which makes using it impossible, so I can't investigate myself at the moment.

closed 2023-10-08 by suborb

th-otto, 2023-10-08:
Maybe you can try using the builds from https://donzé.ch/atari/software/macaranym/

They are more recent than last official builds, and done by an actual macOS user (i'm not ;))

suborb, 2023-10-08:
Perfect - a working keyboard! And I now know that the nightly binaries work!

So on my ancient main machine using Macaranym and working on adv_a.c (about 3000 lines):

`zcc +test -c adv_c.c`

[image]

`zcc +test adv_a.o`

[image]

Which isn't showing the reported issue, the MSX library is a bit larger which adds a couple of seconds to the link time but again, I don't think that's the cause of the reported issue.

Just for reference, an md5sum on the z88dk kit (about 64 meg) yields:

[image]

On the actual machine itself:

```
% time zcc +test -c adv_a.c
zcc +test -c adv_a.c  1.75s user 0.14s system 94% cpu 1.991 total
% time zcc +test adv_a.o
zcc +test adv_a.o  0.79s user 0.06s system 94% cpu 0.901 total
```

and md5sum:

```
md5 z88dk-atari-mint-latest.zip  0.19s user 0.03s system 103% cpu 0.216 total
```

For lack of anything else, let's use a ratio of the md5sums (the assumption here is that we're CPU bound).  Say my host is 70 times quicker than the emulator. In that the compilation time is "about right", but the linker time is actually significantly shorter than would be expected.

th-otto, 2023-10-08:
Are you using the hostfs of aranym? That is, are you accessing the file on the host from inside aranym when compiling linking? That might explain the differences between compiler/linker. I/O is on the host in that case.

Of course generally, aranym isn't a good choice when comparing benchmarks. Even on older machines it is magnitudes faster than anything that was available in real hardware. And the JIT version is about 10x faster again than the normal version.

suborb, 2023-10-08:
Yeah. It’s not ideal. I was hoping there would be an obviously lengthy time spent linking but there isn’t. 

This one is going to be tricky to pin down without real hardware. The lib files aren’t that large so even constantly seeking in them (if we do that) shouldn’t be too painful even for a fat file system.