z88dk/z88dk #2402: Add support for KC160 (KL5C160xxx)

suborb, 2023-10-03:
References to this chip are quite elusive and I've not tracked down a data sheet/manual etc.

I've found a link to the LSI compiler which has an example of the code: https://web.archive.org/web/20220729103236/https://www.lsi-j.co.jp/official/wp-content/uploads/2019/02/lsic80_AnEnglishPamphlet.pdf

So it looks like fairly sensible extensions - just need to find the data sheet and see what it's like.

suborb, 2023-10-03:
Spec: https://web.archive.org/web/20040617191244/http://www.k-micro.com:80/img/seihin/pdf/kcg309911a.pdf

A few more turn up with a search for KL5C16030 on archive.org for pdfs

Root page: https://web.archive.org/web/20021002073108fw_/http://www.k-micro.com/seihin/micon.html and onto: https://web.archive.org/web/20020812204016fw_/http://www.k-micro.com/seihin/micon_kc160doc.html

and bingo: https://web.archive.org/web/20040805150024fw_/http://www.k-micro.com/img/seihin/pdf/KC160inst_v10.pdf

suborb, 2023-10-03:
There's these additional prefixes:

[image]

Which take over the ld r,r as we've seen before and they're used to access 24 bit memory, using a 16 bit operand extended by an 8 bit register.

Mnemonics are a bit weird at a first glance:

[image]

I guess this is `ld (X0x8000),de` or `ld X,(0x8000),de`? Sits slightly oddly with `LDF` used with a 24 bit address constant:

[image]

suborb, 2023-10-04:
As far as I understand the CPU has two modes:

- Z80 "compatibility box" which runs in 0x000000 -> 0x00ffff and has a ROM/RAM split 
- Expanded mode for the rest of the address space: 0x0100000 -> 0xffffff with registers being extended by the XP,YP,ZP,a 8 bit registers.

In expanded mode there is a code/data split, so self-modifying code isn't possible (i.e. the data address space is different to the instruction address space) - the Rabbit has the same feature.

I _think_ the mode is determined by the current (P)PC counter, so if you jump out of the z80 mode, then 24 bit registers come into play.

In compatibility mode the `ld r,r` prefixed instructions do not function. But you can access the upper 8 bit registers using the instructions in the ED page. It's possible to access far memory using the `ldf` instructions. 

The `add/adc/sbc Xhl,rr` sign extended out `rr` to 24 bits.

Attached is a spreadsheet of my workings with the new ED opcodes mapped out and the `ld r,r` prefixed instructions marked.

[kc160_new_inst.xlsx](https://github.com/z88dk/z88dk/files/12800918/kc160_new_inst.xlsx)

pauloscustodio, 2023-10-07:
z80asm finished and merged. Let me know of any issues.

suborb, 2023-10-07:
Thanks @pauloscustodio - it's really appreciated.

 I'd like your thoughts on this: https://github.com/z88dk/z88dk/commit/bcf92e0578b319879887d180b53a18c4b29b88a5#r129364604 - see if you come to the same understanding as me - it might mean a few tweaks to both assembler and disassembler.

suborb, 2023-10-08:
To wrap up and close this issue (and a reminder for the future)

- z80asm/dis: `-mkc160` can use the prefixes for the segment registers
- z80asm/dis: `-mkc160_z80` can't use the prefixes, is compatible with z80_strict, 8080

The library is using -mkc160

closed 2023-10-08 by suborb