z88dk/z88dk #2404: Warnings overflow in conversion from int to uint8_t

th-otto, 2023-10-04:
At several places in ticks.c i get compile warnings:
https://github.com/z88dk/z88dk/blob/e30ddfdee33266d514902be3bc16f01e20443315/src/ticks/ticks.c#L4163
https://github.com/z88dk/z88dk/blob/e30ddfdee33266d514902be3bc16f01e20443315/src/ticks/ticks.c#L4242
https://github.com/z88dk/z88dk/blob/e30ddfdee33266d514902be3bc16f01e20443315/src/ticks/ticks.c#L4333

I don't know yet what that code does, but i guess that should be ``& 255`` ;)

suborb, 2023-10-04:
Ah, that's me being stupid - it's meant to be a boolean yes/no. Odd that I didn't get a warning about it.

th-otto, 2023-10-04:
Maybe just different compiler version? I was using gcc 9.5.0

suborb, 2023-10-05:
Closed via #2405

closed 2023-10-05 by suborb