z88dk/z88dk #2408: math - add r4k & kc160 - specifically mulu_32_16x16 optimisation

feilipu, 2023-10-07:
- [ ] Integer routines (used by math16 too)
- [ ] math32 multiply & sqr routines

Derived from r2ka solution, so quick fix.

suborb, 2023-10-08:
The r2ka code works on the r4k unless, so unless you want to take advantage of the mulu instruction there may not be much to do there.

feilipu, 2023-10-08:
The main thing to do is to replace the (slow) unsigned multiply 32_16x16 function used by r2ka with the r4k aunsigned opcode. And comb the code again to use r4k loads and shifts (though I've done this one already).

For kc160, it would default to using z80 functions which uses none of the hardware opcodes. But IDK whether it is worth the effort. If there are actual users of the CPU, then I guess, yes. For math16 the unsigned 32_16x16 is the critical optimisation.

I have a few days at end of Oct, otherwise just to remember to do it in Dec.

suborb, 2023-10-15:
> **Edit** Also, wouldn't the r2ka opcodes not be available in r4k mode? Specifically r4k provides a `mulu` but no `muls` opcode, iirc.

4k mode includes `muls` - the switch toggles on the 0x6d, 0x7f pages and the new instructions in the main page.

feilipu, 2023-10-15:
This is done, once you commit the [kc160_mult](https://github.com/z88dk/z88dk/compare/kc160_mult) branch.

feilipu, 2023-10-15:
Noted that for mbf32 the r2ka math test result is substantially better than for r4k.
So there's some optimisation code being missed out.

suborb, 2023-10-15:
> Noted that for mbf32 the r2ka math test result is substantially different and better than it is for r4k. So there's some optimisation code being missed out, but I can't see why.

The middle bit of the mainpage (`ld r1,r2` and the  ALU instructions - `add`, `add`, `sub`, `sbc`, `and`, `xor`, `or`, `cp`) were moved to the 0x7f page so the time taken for them doubled from 2T to 4T

suborb, 2023-10-15:
All done, I fixed up some slightly wonky CPU detection in mbf32 as well

closed 2023-10-15 by suborb