z88dk/z88dk #2410: Could somebody help me with compiling for the ti84+

HeronErin, 2023-10-10:
Hello,

Currently I am writing [TiConstructor](https://github.com/HeronErin/TiConstructor), a project for making programs/apps for the ti-84+ (non ce), and have been using the normal sdcc for compiling with my own crt0 files. But have been noticing that sdcc generates some pretty bad assembly, has some issues with syntax (ie struct->attr+1 not returning the incremented attribute, but instead the normal value), and overall a quite bad slandered library.  So am working on transitioning to using zcc as the compiler.

However have not been able to get the calculator to accept anything generated by zcc, and at best giving me a "ERR: INVALID".

The following is a bare bones helloworld I am using for testing that should compile for both zcc and sdcc, and produces a valid 8xp with my old build script:
 ```c
#ifdef SDCC
#define bcall(__LABEL__) rst	40 \
                        .dw __LABEL__
// A bit of trash data so the program runs correctly
void _(){__asm ret
	     .ascii "0000000000" __endasm;}

#else
#pragma string name ad
#define bcall(__LABEL__) rst 40 \ defw __LABEL__

// A bit of trash data so the program runs correctly
void _(){
	#asm
	ret
	defm "0000000000"
	#endasm
}
#endif

void main()
{

	#ifdef SDCC
	__asm
	#else
	#asm
	#endif
		rst 40
		bcall(0x4540 ) ; _ClrLCDFull

		xor	a, a
		ld	(0x86D7), a
		ld	(0x86D8), a

		#ifdef SDCC
		ld a, #'!'
		#else
		ld a, '!'
		#endif

		push	ix
		bcall(0x455E)     ; _VPutMap
		pop	ix

		bcall(0x4972 )    ; _getkey

	#ifdef SDCC
	__endasm;
	#else
	#endasm
	#endif
	

}
```
And here is the build script that works to compile this for sdcc:
```bash
DIRECTORY=$(cd `dirname $0` && pwd)

cd x

export MAINC="../hello.c"
export OUT_NAME="testtex"

sdasz80 -p -g -o tios_crt0.rel $DIRECTORY/other_files/tios_crt0.s

sdcc -DSDCC --no-std-crt0 --code-loc 40347 --data-loc 0 --std-sdcc99 -mz80 --reserve-regs-iy -o $OUT_NAME.ihx tios_crt0.rel $MAINC

objcopy -Iihex -Obinary $OUT_NAME.ihx $OUT_NAME.bin
python $DIRECTORY/other_files/binpac8x.py $OUT_NAME.bin -O $OUT_NAME

rm $OUT_NAME.bin $OUT_NAME.ihx $OUT_NAME.lk $OUT_NAME.lst $OUT_NAME.map $OUT_NAME.noi $OUT_NAME.rel $OUT_NAME.sym tios_crt0.rel
```

If plausible I would like to adapt the build script above for zcc (ie compile a .bin that binpac8x and rabbitsign can use), as in my [build_app.sh](https://github.com/HeronErin/TiConstructor/blob/master/build_app.sh) file I use rabbitsign to generate an app for the ti-84, so that any program written with my library can be compiled as both an app and a program. However have had little success in getting a program to compile and run on the calculator when using zcc. 

I am using the following command to attempt to compile
`docker run -v ${DIRECTORY}:/src/ z88dk/z88dk zcc +ti83p -lm -o x/v.bin hello.c -create-app`

HeronErin, 2023-10-10:
And `docker run -v ${DIRECTORY}:/src/ z88dk/z88dk zcc +ti83p -subtype=asm -lm -o x/v.bin hello.c -create-app` also gives the  same error

suborb, 2023-10-11:
I'm not a TI person (it's really the only target I've never run), however for the TIs `-create-app` effectively runs `bin2var` and  generates a `.ti8*` file, so I think `-create-app` isn't useful for you.

Fabrizio-Caruso, 2023-10-11:
Building for the TI calculators is not particularly complex if target one of the supported ones (I fear the TI84+ may not supported in Z88DK unless it is compatible with the TI85 or TI83). What I find really hard and cumbersome is the load procedure on the emulated machine. I have tried the real hardware.

I have done it in Cross-Lib.
For instance I build my games for the TI85 with:
https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/src/games/chase/makefiles.chase/makefiles_z88dk/Makefile_z88dk_q-z#L736

which I managed to load into an emulator with these steps:
https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/docs/HOW_TO_LOAD_THE_GAME.md#ti-85

HeronErin, 2023-10-12:
By my understanding ti-83 plus should be compatible with the ti-84 plus. My issue was been that `-create-app`  can not be loaded into jstified, and when using binpac8x with the .bin I get an invalid error, next what I think I will do is compare the outputted .bins in the hex editor to see what can be done (possibly to create a python script to "repair" the 8xp)

HeronErin, 2023-10-12:
Update:
After reading through #1622 I learned that the error comes from the asm program not starting with the AsmPrgm token (BBh,6Dh) meaning, and after you just need to prepend it to the .bin for the program to be accepted by the calc. Might need to write a PR to fix it. 
```bash
docker run -v ${DIRECTORY}:/src/ z88dk/z88dk zcc +ti83p -subtype=asm -startup=10 -Cz--altfmt -Cz--comment=what -o x/hello.bin hello.c
cat <( echo -ne "\xbb\x6d" ) x/hello.bin > x/x.bin
cd x
python $DIRECTORY/other_files/binpac8x.py x.bin -O ad
```

Fabrizio-Caruso, 2023-10-13:
@HeronErin are you also using an emulator to try your programs?
If yes, could you please tell me which one and how do you manage to load and run your home-brews in the emulator?
If you figure out how to load and run your programs, would you please share your scripts so that others can use them?
I have written 8 games that should run on all Z88DK_supported TI calculators but the load procedure was working randomly last time I tried to load them into an emulator.

HeronErin, 2023-10-13:
@Fabrizio-Caruso I am using [jstified](https://www.cemetech.net/projects/jstified/) and I was able to use my own scripts to get both apps and programs compiling. The first issue was that `-create-app` just does not work, so you must do it yourself.  I used binpac8x.py and rabbitsign in the end.  The second issue was that the crt0 file did not contain the AsmPrgm token, so I just add it with echo. The third issue was that flash app compilation was never fully implemented, and the existing crt0 file is years out of date, so I had to rewrite it based on the crt0 file for ram programs. [Here](https://github.com/HeronErin/TiConstructor/blob/bd6dbbbec408dc007798ab81680bba8696cf0f9e/build_prog.sh) you can find my build script for making ram programs. And [here](https://github.com/HeronErin/TiConstructor/blob/bd6dbbbec408dc007798ab81680bba8696cf0f9e/build_app.sh) is the build script for flash apps. And [here](https://github.com/HeronErin/TiConstructor/blob/bd6dbbbec408dc007798ab81680bba8696cf0f9e/other_files/ti83p_crt0_app.asm) is my crt0 file for apps (but it might break over parts of the library, I just tested a small hello world app)

Fabrizio-Caruso, 2023-10-18:
@zx70 I now think that I probably never managed to get TI83+ binaries to load into an emulator (because I have documented and written complex instructions on how to get ti82, ti83, ti85, ti86 to work and not for the ti83+). So, I wonder whether appmake is creating something usable on any TI83+ emulator.

zx70, 2023-10-18:
TBH I think that the TI83 Plus support is not very bad.
The real problem is the different binary formats used by tools and emulators,  plus the different editions (TI84 etc).   The TI calculators were developed and tested on the VTI emulato, but there are more options to add, probably mode than the ones I'm aware of. 
Consider that many of the original ti83plus shells were already able to import and run the corresponding ti83 programs, it is a tempting quick win.

HeronErin, 2023-10-19:
Ya, it is not supported that well, and I am needing to use many of my own functions since the standard library doesn't work well with flash apps, but it is still the single best way to develop with c for the ti84, and the only c alternative is sdcc,  but that has no optimization. I'm not familiar enough with z88dk to try to fix it without unknowingly breaking some other system I didn't even know existed, so think I will just fix the crt0 files, interrupts, and write my own functions for my TiConstructor project, which as far as I can tell is the best c toolchain for the ti83/83p. I hope the ti84/83p gets better support but since it is the least popular of the many target systems I can understand it not being a priority.

zx70, 2023-10-19:
Could you share the changed stuff?   I'd really like to carbon copy your code and make it become a parallel option.
In the worst case it could become a totally new target setup and maybe get merged later.

HeronErin, 2023-10-19:
@zx70 Sure thing, everything I am doing is in the [z88dk-compiler-migration](https://github.com/HeronErin/TiConstructor/tree/z88dk-compiler-migration) branch of the [TiConstructor](https://github.com/HeronErin/TiConstructor) project, It is far from done, but it is coming along.  Right now I am just modifying the .bin file and using binpac8x for ram programs. And a custom crt0 for flash apps then signing it with rabbitsign. But I want to start looking at the code for Z88DK-APPMAKE and compare it to binpac8x to see what might be corrupting the 8xp files. And want to see if I can add functionality to allow for signing flash apps straight from appmake instead of relaying another program.

zx70, 2023-10-20:
Well, at a first glance we have already an interesting difference on the TI84 allowing to run reasonably big program on a barebones asm() TI call, which is not common at all.
I hit the binary format limit so many times that I'm a bit discouraged, but I'll try to follow a straight TI84 emulation path and see if I'm able to spot what's missing on appmake first, which I understand to be a common goal.
The 'TI shells' world is something I'd like to keep supporting because it's how things must be done on the earlier models and allows a nice portability between different ROM versions.  NoShell didn't exist at the time, probably it must be in the TODO list.

HeronErin, 2023-10-20:
The file size limit is the reason that I think supporting flash apps is important, they don't have a limit, and the 83/84 supports multi page apps. Although multi page apps would require changes to the compiler since you need to use bcall instead of call for out of page functions. I can already get single page apps working with rabbitsign, but puts() doesn't seem to work. So I'm not sure if I'm just missing something in my crt0 or I would need to make changes to the library itself. (It should also be noted that you can't do self modification in flash apps, and that is why global non constant variables don't work in sdcc, so I'm not sure if that could break anything). Honesty I'm not sure if it would be better to reimplment the app signing or just sub module rabbitsign as it is open source(gpl) and works well. (Also supporting other calc models, but not sure how well)

zx70, 2023-10-20:
I'm the weakest reference in the z88dk team for paging, but we have already a good base to get it working on other targets.
Part of the problem is on the packaging, once the compiler spit the code into sections they must be preformatted and loaded correctly by the target system. 
z88dk is a good choice to deal with this kind of stuff because the whole toolchain is under our control, but this doesn't mean that code paging is an easy achievemt.   An intermediate result is the FAR memory functions set.  It is part of the z88dk kit since the beginning and often forgotten but being able to set/get data from another memory page can be useful already.

zx70, 2023-10-20:
..in the meantime in the tutorial you are mentioning in your pages I spotted SPASM.
This file could help a little:  https://github.com/alberthdev/spasm-ng/blob/master/export.cpp

zx70, 2023-10-21:
Good, I chose to keep this new format and hide the old one behind a parameter on appmake.

'alftmt' does not exist anymore, it is the default behaviour.  A new 'oldfmt' option wraps a package in the old way, for some of the ancient tools or early VTI releases.

HeronErin, 2023-10-21:
Woo, hope this doesn't break somebody's build chain or something lol. (XKCD 1172 and all that) But good to see good things happening

zx70, 2023-10-21:
it's happening because of you  <grin> 
honest,  it was  your introductory pages putting me I  the right direction.   
As said, the existing build chains can restore the previous behaviour by using --oldfmt on appmake

HeronErin, 2023-10-24:
@zx70 Hey, I have managed to modify z88dk to get flash apps compiling. [Here](https://github.com/HeronErin/z88dk/commit/92a086faee3b93137d8907cbe5e084131850041a) is a link to the commit on my github repo. I have copied a large amount of code from spasm, but have got it working with zcc, and my example code from the first comment runs correctly in jstified.  I am not sure if it is ready yet as puts() in "stdio.h" does not work. Also multi page apps have not been tested.  You can compile with `zcc +ti8xk main.c -o x.bin -create-app -Cz"-n xtest"`

I am still inexperienced with z88dk, and want to know if there are anything that you think can't be added. Appmake's make file has `-lgmp -lssl -lcrypto`  added for the crypto stuff needed for signing apps and I was unsure if adding dependencies was not a good idea.

zx70, 2023-10-25:
It's very good to have a separate C file for appmake, if you don't mind I'd begin incorporating your work :)

HeronErin, 2023-10-25:
I am planing on making a PR after I am finished with support for multi page apps, as of now my plan is to compile each page separately and combine them with appmake. After reading the infomation on multi page apps on the official [developer guide](https://ia800606.us.archive.org/16/items/83psdk/sdk83pguide.pdf) I have learned that the only way to interact with otherpages is bcall() and bjump() so it makes more sense to link them together later. However am not to sure the best way to do it. As of now I am trying to do it with multiple custom #pragmas and exporting a .map from the compiler, but this seems like a hacky way to do it. 

Ideally I would have some sort of function decorator that would tell the compiler to export the symbol for later.
And for other pages to replace a `call _function`  with a `bcall(_function)` (where bcall is not the function label but a location in a branch table that is after the app header)
But don't know how to do this (or if something like this already exists)

Do you know if there is already a way to overide the calling behavior in z88dk? Or atleast a way to better export functions?

suborb, 2023-10-25:
There's a few ways to deal with long calls. The one that's had the most use is `__banked` - this requires the writing of banked_call trampoline.

However, the closest to the TI calc calling convention is `__z88dk_shortcall(RR, VV)`.

In both cases, you'll have to build the vector table yourself since we've not got any magic to do that.

Some pages:

* https://github.com/z88dk/z88dk/wiki/More-Than-64k
* https://github.com/z88dk/z88dk/wiki/CallingConventions#banked-calling-convention-modifiers

PR #2433 (2023-11-13): Added support for ti83p/ti84p flash apps.
I have added support to building both single-page and multi-page flash apps for the ti83p/ti84p. A lot of things had to be added, so feel free to look through and see if I have left any bugs, or better yet tell me a better way of doing it.  The crypto stuff for signing apps was mostly taken from [spasm-ng](https://github.com/alberthdev/spasm-ng) and the rest I wrote. 

It should be noted that at this point puts() is not functioning for flash apps, but fputc_cons() is, so something needs fixed in the standard library somewhere. 

With the recent changes [... cut]
closed 2023-10-27 by HeronErin