z88dk/z88dk #2411: [psg][joystick?][cpc][bug] Using PSG lib breaks joystick API

Fabrizio-Caruso, 2023-10-11:
This issue has been reproduce on several emulators (e.g., RetroVirtualMachine) by me and @X3NoMoRPH (https://github.com/X3NoMoRPH) and on *real* hardware only by @X3NoMoRPH. This issue is not detected by some weak emulators such as CaPriCe.

The test on real hardware was done with my latest Cross-Lib game stinger.
(reproducible with the disk image produced in Cross-Lib by running `xl stinger cpc`)

The minimal example (with simplified code excerpts from my game) below reproduces the very same issue:
```
// CPC: zcc +cpc  test_joy2.c -subtype=dsk -lndos -otestjoy2.dsk -create-app

#include <stdio.h>
#include <time.h>
#include <unistd.h>
#include <stdlib.h> 
#include <sound.h>

#include <games.h>
#define YSize 20

#include <psg.h>
#include <stdint.h>
#define A_PERIOD_LOW 0
#define A_PERIOD_HI 1
#define B_PERIOD_LOW 2
#define B_PERIOD_HI 3
#define C_PERIOD_LOW 4
#define C_PERIOD_HI 5
#define NOISE 6
#define CONTROL 7
#define A_VOLUME 8
#define B_VOLUME 9
#define C_VOLUME 10  
#define ENV_PERIOD_LOW 11
#define ENV_PERIOD_HI 12
#define ENV_WAVE 13

#define POKE(addr,val)     (*(uint8_t*) (addr) = (val))

int input;
int stick = 1;
int time_counter;
int fire_pressed = 0;

void _XL_ZAP_SOUND(void) 
{
    uint8_t i;
    uint8_t j;
    set_psg(A_VOLUME,15);
    set_psg(A_PERIOD_LOW,255);  
    set_psg(A_PERIOD_HI, 15);  
    set_psg(CONTROL, 0xFF  - 0x01);  
    for(i=0;i<16;i++)
    {
        set_psg(A_PERIOD_HI,15-i);        
        for(j=0;j<150U;++j)
        {
        }
    }    
    set_psg(A_VOLUME,0);        
}

uint8_t fire_pressed_after_time(void)
{
    return (!time_counter) && (fire_pressed = (joystick(stick)==MOVE_FIRE));
}

void short_sleep(void)
{
    uint16_t i;
    for(i=0;i<200;++i)
    {
    }
}

int main(void)
{
    input = 0;
    uint8_t i;
    time_counter = 5;
    do
    {
        for(i=3;i<(YSize-5)*4;++i)
        {
            if(time_counter)
            {
                --time_counter;
            }            
            // Do something here
            if(fire_pressed_after_time())
            {
                break;
            } 
            // Do something else here
            short_sleep();
            if(fire_pressed_after_time())
            {
                break;
            }
        }
        if(!fire_pressed)
        {
            // Do something
        }
    }
    while(!fire_pressed);    
    printf("OK1");
	
	// **********************************
    _XL_ZAP_SOUND(); // **** BUG HERE *** 
    // **********************************
	
	sleep(1);
    time_counter = 5;
    do
    {
        for(i=5;i<YSize-2;++i)
        {
            if(fire_pressed_after_time())
            {
                break;
            }
            if(time_counter)
            {
                --time_counter;
            }
        }
    }
    while(!fire_pressed); 
    printf("OK2");
    while(1){};
    return 0;
}
```

which I build with
```
zcc +cpc  test_joy2.c -subtype=dsk -lndos -otestjoy2.dsk -create-app
```

suborb, 2023-10-11:
The CPC is such a miserable cranky machine to do anything with.

The joysticks are scanned with interrupts disabled, but `set_psg()` doesn't disable the interrupts so the firmware is probably crapping on us from a great height as it tends to do.

If you surround the call to `_XL_ZAP_SOUND()` with `asm("di"); _XL_ZAP_SOUND(); asm("ei");` does that solve the problem?

Fabrizio-Caruso, 2023-10-11:
Yes ! Thanks!
`set_psg(CONTROL, 0xFF  - 0x01-0x40);` seems to fix the issue!
Once I rebuild the game I would kindly ask @X3NoMoRPH to retest on real hardware.

Fabrizio-Caruso, 2023-10-11:
@suborb 
Would `set_psg(CONTROL, 0xFF  -0x41);` be safe on all PSG targets?

suborb, 2023-10-11:
> @suborb
> Would `set_psg(CONTROL, 0xFF -0x41);` be safe on all PSG targets?

The MSX also needs the PORTA bit to be 0. However +zx uses PORTA for the rs232 so I don't think it's safe.

For the CPC I can patch `set_psg()` to ensure that bit is reset - the MSX version already does that.

Fabrizio-Caruso, 2023-10-11:
@suborb So for the moment I use an conditional macro only for the CPC but I should be able to remove in the next days/weeks?

suborb, 2023-10-11:
Yes, next time you upgrade you can remove the conditional macro - the next nightly should have the patch.

Fabrizio-Caruso, 2023-10-11:
@suborb Thanks a lot for your support!

closed 2023-10-11 by Fabrizio-Caruso