z88dk/z88dk #2416: understanding ez80 assembly suffixes

feilipu, 2023-10-15:
I am looking at some r2ka ez80 code to convert some `defb` into correct suffixes and opcodes, and I'm seeing something I don't understand.

For example [`lp.gchar.asm`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/l/sccz80/4-ez80/lp_gchar.asm) has
```asm
    defb    $5b    ;lil
    push    hl
    defb    $5b    ;lil
    ld      hl,2
    defb    0
    defb    $5b
    add     hl,sp
    defb    $5b
    ld      (hl),e
    defb    $5b
    pop     hl
    defb    $5b
    ld      a,(hl)
    ret
```

With the first `defb $5b ;lil`    `push hl` the implication is the assembled hex should be `$5B $E5`.
But, that sequence doesn't exist according to the datasheet.
The options in the datasheet are `push` `$E5`, `push.s` `$52 $E5`, or `push.l` `$49 $E5`.

This is `$5B` is repeated throughout the `lp_*` functions, although many of the opcodes offer only `$52` `*.s` or `$49` `*.l` suffix options.

If the `$5B` `lil` is a persistent mode change, shouldn't appear only once?

Or, can `ASSUME ADL = x` be inserted only once?

So, I'm not sure what to make of this?

suborb, 2023-10-15:
To be honest, I was very confused about the suffixes when writing that, and referring to code that was similarly very confused so I would't be surprised if it was wrong.

The idea behind this one was to push `hlu` onto `spl`, nobble bits 16-24 and then pop `hlu` pointing to the far address.

The suffix byte only applies to the attached opcode

feilipu, 2023-10-15:
I think Paulo has done `ASSUME ADL=`, so I imagine that's where all the datasheet examples start, by setting ADL.

Then the opcodes' suffix (or precursor byte) can determine exceptions to that assumption.

But I didn't understand it properly either. Hence the questions...

Paulo's opcode lists do reflect the datasheet description bytes though, so the current binary outcome can't be reproduced by using natural opcodes.

pauloscustodio, 2023-10-15:
Yes, you can type ```.assume adl=0``` or ```.assume adl=1```

feilipu, 2023-10-16:
Some reference notes from the datasheet on how this is supposed to work, to help me get my head around it.

## MBASE Register

The MBASE register can only be changed while in ADL mode, thereby preventing accidental page switching when operating in Z80 MEMORY mode. The MBASE address register does not affect the length of the CPU register. In Z80 mode, the CPU registers remain 16 bits, independent of the value of MBASE.

When ADL mode is selected, MBASE does not affect memory addressing.

In ADL mode, all addresses and data are 24 bits. All data READ and WRITE operations pass 3 bytes of data to and from the CPU when operating in ADL mode (as opposed to only 2 bytes of data while in Z80 mode operation). Although MBASE does not affect operation during ADL mode, the MBASE register can only be written to when operating in ADL mode.

## PC Register

The Program Counter (__PC__) is automatically incremented during program execution. When a program jump occurs, the new value is placed in the Program Counter, overriding the incremented value. In Z80 mode, the Program Counter is only 16 bits; however, a full 24-bit address `{MBASE,PC[15:0]}`, is used. In ADL mode, the Program Counter is returned by `{PC[23:0]}`.

## SP Register

Stack Pointer Long (__SPL__) in ADL mode, the 24-bit Stack Pointer Long stores the address for the current top of the external stack. In ADL mode, the stack can be located anywhere in memory. The external stack is organised as a last-in first-out (LIFO) file. Data can be pushed onto the stack or popped off of the stack using the PUSH and POP instructions. Interrupts, traps, calls, and returns also employ the stack.

Stack Pointer Short register (__SPS__) in Z80 mode, the 16-bit Stack Pointer Short stores the address for the current top of the stack. In Z80 mode, the stack can be located anywhere within the current Z80 memory page. The current Z80 memory page is selected by the MBASE register. The 24-bit Stack Pointer address in Z80 mode is `{MBASE, SPS}`. The stack is organised as a last-in first-out (LIFO) file. Data can be pushed onto the stack or popped off of the stack using the PUSH and POP instructions. Interrupts, traps, calls, and returns also employ the stack.

## Control Bits

Address and Data Long Mode Bit (__ADL__) the ADL mode bit indicates the current memory mode of the CPU. An ADL mode bit reset to 0 indicates that the CPU is operating in Z80 MEMORY mode with 16-bit Z80-style addresses offset by the 8-bit MBASE register. An ADL mode bit set to 1 indicates that the CPU is operating in ADL mode with 24-bit linear addressing. The default for the ADL mode bit is reset (cleared to 0). The ADL mode bit can only be changed by those instructions that allow persistent memory mode changes, interrupts, and traps. The ADL mode bit cannot be directly written to.

Mixed-ADL Bit (__MADL__) the MADL control bit is used to configure the CPU to execute programs containing code that uses both ADL and Z80 MEMORY modes.

## CPU Registers in Z80 Mode

The Program Counter register (PC) is 16 bits long. The address being executed is 24 bits long and is composed as `{MBASE, ADDR[15:0]}`. While the MBASE register is only used during Z80 mode operations, it cannot be written while operating in this mode.

In Z80 mode, the upper byte (bits 23:16) of each multi-byte register is undefined. When performing 16-bit operations with these registers, the application program cannot assume values or behaviour for the upper byte. The upper byte is only valid in ADL mode.

## CPU Registers in ADL Mode

In ADL mode, the BC, DE, HL, IX and IY registers are 24 bits long for multibyte operations and indirect addressing.

None of the upper bytes (BCU, DEU, HLU, IXU, IYU, etc.) are individually accessible as standalone 8-bit registers.

MBASE is not used for address generation in ADL mode; however, it can only be written in ADL mode.

The Program Counter is 24 bits long, as is SPL. IEF1, IEF2, ADL, and MADL are single bit flags.

# Memory Mode Switching

The CPU is capable of easily switching between the two available memory modes (ADL mode and Z80 mode). There are two types of mode changes available to the CPU: persistent and single-instruction. For example, persistent mode switches allow the CPU to operate indefinitely in ADL mode, then switch to Z80 mode to run a section of Z80 code, and then return to ADL mode. Conversely, single-instruction mode changes allow certain instructions to operate using either addressing mode without making a persistent change to the mode.

## Mode Compiler Directives

In the Zilog ZMASM/ZDS assembler, the application code is assembled for a given state of the ADL mode bit by placing one of the two following compiler directives at the top of the code:
`.ASSUME ADL = 1`
`.ASSUME ADL = 0`

These directives indicate that either ADL MEMORY mode (ADL = 1) or Z80 MEMORY mode (ADL = 0) is the default memory mode for the code being currently compiled.

The assembler doesn't set either mode, but interprets the following instructions according to the noted mode. The code developer is responsible for ensuring that this source file setting matches the actual state of the hardware ADL mode bit when the code is executed.

## Opcode Suffixes for Mode Control

There are four individual suffixes available for use: `.SIS`, `.SIL`, `.LIS`, and `.LIL`. These suffixes are appended to many instructions to indicate that a memory mode change or an exception to standard memory mode operation is being requ [... cut ...]
The Instruction Stream Short and Instruction Stream Long suffixes, `.IS` and `.IL`, control whether a multi-byte immediate data or address value fetched during instruction execution is 2 or 3 bytes long (for example, a LD HL, Mmn instruction versus a LD HL, mn instruction). The CPU must know whether to fetch 3 bytes (Mmn) or 2 bytes (mn) of data. The `.IS` and `.IL` portions of the suffix tell the CPU the length of the instruction.

If the length of the instruction is unambiguous, the `.IS` and `.IL` suffixes yield no effect.

feilipu, 2023-10-16:
So, I think that the `lp_ghar.asm` could look like this code... for discussion.

```asm
;
; Entry: EHL=far pointer
; Exit: a=byte

lp_gchar:
    push.l    hl       ; push 3 bytes (HLU is possibly undefined depending on mode)
    ld.il     hl,2     ; point to pushed HLU by adding a 24 bit value to SPL
    add.l     hl,sp
    ld.l      (hl),e   ; store the E register at the location in SPL
    pop.l     hl       ; get a correct long address in 24 bit HL
    ld.l      a,(hl)   ; get a byte from that location into A.
    ret.l              ; return correctly from a long call
```

feilipu, 2023-10-17:
See #2424.