z88dk/z88dk #2417: (z80asm) Issue with spaces in the directory name

suborb, 2023-10-15:
Reported in the forum here: https://z88dk.org/forum/viewtopic.php?f=13&t=11849

I've reproduced by nobbling zcc:

```
--- a/src/zcc/zcc.c
+++ b/src/zcc/zcc.c
@@ -3259,7 +3259,7 @@ void tempname(char *filen)
                                         * files. */
         *ptr = 0;    /* Don't want to risk too long filenames */
 #else
-    strcpy(filen, "/tmp/tmpXXXXXXXX");
+    strcpy(filen, "/tmp/with space/tmpXXXXXXXX");

     /* Prevent linker warning: the use of mktemp is dangerous */
     /* mktemp(filen);                                         */
```

Running:

```
zcc +test x.c y.c -v

PROCESSING x.c
z88dk-ucpp -iquote"." -D__TESTTARGET__ -DZ80 -D__Z88DK  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/Users/dom/z88dk/lib/config/../..//include"    "x.c" "/tmp/with space/tmpEPuUB3JK.i2"
z88dk-zpragma -sccz80 -zcc-opt="/tmp/tmpzcc7xEiaDok/zcc_opt.def" < "/tmp/with space/tmpEPuUB3JK.i2" > "/tmp/with space/tmpEPuUB3JK.i"
z88dk-sccz80  -ext=opt -mz80 -zcc-opt="/tmp/tmpzcc7xEiaDok/zcc_opt.def" -standard-escape-chars   "/tmp/with space/tmpEPuUB3JK.i" -o "/tmp/with space/tmpEPuUB3JK.opt"
z88dk-copt -mz80  "/Users/dom/z88dk/lib/config/../..//lib/z80rules.9" < "/tmp/with space/tmpEPuUB3JK.opt" > "/tmp/with space/tmpEPuUB3JK.op1"
z88dk-copt -mz80  "/Users/dom/z88dk/lib/config/../..//lib/z80rules.2" < "/tmp/with space/tmpEPuUB3JK.op1" > "/tmp/with space/tmpEPuUB3JK.opt"
z88dk-copt -mz80  "/Users/dom/z88dk/lib/config/../..//lib/z80rules.1" < "/tmp/with space/tmpEPuUB3JK.opt" > "/tmp/with space/tmpEPuUB3JK.asm"
cat "/tmp/with space/tmpEPuUB3JK.asm" >> "/tmp/with space/tmpEPuUB3JK.tmp"
cat "/tmp/with space/tmpEPuUB3JK.tmp" >> "/tmp/with space/tmpEPuUB3JK.asm"
z88dk-z80asm -I"/Users/dom/z88dk/lib/config/../..//lib/target/test/def" -I"/tmp/tmpzcc7xEiaDok" -D__SCCZ80  -s  -mz80 -I"/Users/dom/z88dk/lib/config/../..//lib" "/tmp/with space/tmpEPuUB3JK.asm"

PROCESSING y.c
z88dk-ucpp -iquote"." -D__TESTTARGET__ -DZ80 -D__Z88DK  -DSCCZ80 -DSMALL_C -D__SCCZ80 -isystem"/Users/dom/z88dk/lib/config/../..//include"    "y.c" "/tmp/with space/tmpCZz7Znxs.i2"
z88dk-zpragma -sccz80 -zcc-opt="/tmp/tmpzcc7xEiaDok/zcc_opt.def" < "/tmp/with space/tmpCZz7Znxs.i2" > "/tmp/with space/tmpCZz7Znxs.i"
z88dk-sccz80  -ext=opt -mz80 -zcc-opt="/tmp/tmpzcc7xEiaDok/zcc_opt.def" -standard-escape-chars   "/tmp/with space/tmpCZz7Znxs.i" -o "/tmp/with space/tmpCZz7Znxs.opt"
z88dk-copt -mz80  "/Users/dom/z88dk/lib/config/../..//lib/z80rules.9" < "/tmp/with space/tmpCZz7Znxs.opt" > "/tmp/with space/tmpCZz7Znxs.op1"
z88dk-copt -mz80  "/Users/dom/z88dk/lib/config/../..//lib/z80rules.2" < "/tmp/with space/tmpCZz7Znxs.op1" > "/tmp/with space/tmpCZz7Znxs.opt"
z88dk-copt -mz80  "/Users/dom/z88dk/lib/config/../..//lib/z80rules.1" < "/tmp/with space/tmpCZz7Znxs.opt" > "/tmp/with space/tmpCZz7Znxs.asm"
cat "/tmp/with space/tmpCZz7Znxs.asm" >> "/tmp/with space/tmpCZz7Znxs.tmp"
cat "/tmp/with space/tmpCZz7Znxs.tmp" >> "/tmp/with space/tmpCZz7Znxs.asm"
z88dk-z80asm -I"/Users/dom/z88dk/lib/config/../..//lib/target/test/def" -I"/tmp/tmpzcc7xEiaDok" -D__SCCZ80  -s  -mz80 -I"/Users/dom/z88dk/lib/config/../..//lib" "/tmp/with space/tmpCZz7Znxs.asm"

PROCESSING /Users/dom/z88dk/lib/config/../..//lib/target/test/classic/test_crt0.asm
cat "/Users/dom/z88dk/lib/config/../..//lib/target/test/classic/test_crt0.asm" >> "/tmp/with space/tmp4K6ecZ2m.asm"
cat "/tmp/with space/tmp4K6ecZ2m.asm" >> "/tmp/with space/tmp4K6ecZ2m.tmp"
cat "/tmp/with space/tmp4K6ecZ2m.tmp" >> "/tmp/with space/tmp4K6ecZ2m.asm"
WILL ACT AS CRT

GENERATING OUTPUT

PROCESSING CRT
z88dk-z80asm -I"/Users/dom/z88dk/lib/config/../..//lib/target/test/def" -I"/tmp/tmpzcc7xEiaDok" -D__SCCZ80  -s  -mz80 -I"/Users/dom/z88dk/lib/config/../..//lib" -I"/Users/dom/z88dk/lib/config/../..//lib/target/test/classic"  "/tmp/with space/tmp4K6ecZ2m.asm"

z88dk-z80asm -mz80 -b -d  -o"a.bin" -L.  -L"/Users/dom/z88dk/lib/config/../..//lib/clibs/z80"   -I"/tmp/tmpzcc7xEiaDok" -D__SCCZ80 -L"/Users/dom/z88dk/lib/config/../..//lib/clibs" -I"/Users/dom/z88dk/lib/config/../..//lib" -ltest_clib -l"z80_crt0"     "@/tmp/with space/tmpOvABI4GV.lst"
/tmp/with space/tmpOvABI4GV.lst:1: error: file not found: /tmp/with
  ^---- /tmp/with space/tmp4K6ecZ2m.o
```

pauloscustodio, 2023-10-16:
Root cause: list files accept filenames and options separated by spaces, and accepts quotes surrounding filenames with spaces.
Easiest solution is to change to accept only one filename or one option per line and accept any spaces within.

suborb, 2023-10-16:
All of our list files have just one file per line, so that seems a sensible solution.

commit 9aa4a8ff (2023-10-16): Fix #2417: accept file names with spaces in list files without quotes
Each list file line can contain only one file or one option.
commit 222b8a51 (2023-10-16): Fix #2417: accept file names with spaces in list files without quotes
Each list file line can contain only one file or one option.
commit bad1204d (2023-10-17): Fix #2417: accept file names with spaces in list files without quotes
Each list file line can contain only one file or one option.
closed 2023-10-17 by pauloscustodio