z88dk/z88dk #2418: (z80asm) Trouble with defc label1 = label2

suborb, 2023-10-15:
Trying to figure out exactly what's going on here. But take a look at stdio/_freopen1.c

All coming out of the example in this post: https://z88dk.org/forum/viewtopic.php?f=13&t=11849 - I'm using +test because it's easier to debug.

There's some jp optimisation going on in _freopen1.c, so there's `defc i_13 = i_9`. Because of the cleanup function, this ends up in the `code_crt_exit` section. When linked this seems to end up point to somewhere in `l_eq` which is "puzzling" since it's supposed to be local.

suborb, 2023-10-15:
Moving the defc into the right section corrects the problem, but I'm confused as to what was actually happening.

pauloscustodio, 2023-10-20:
> Moving the defc into the right section corrects the problem, but I'm confused as to what was actually happening.

The copied symbol was in a different section, so the address base was different on relocation. Fixed.

commit 5ad03323 (2023-10-20): Fix #2418: move DEFC alias to section of copied symbol
commit 8608e80a (2023-10-20): Fix #2418: move DEFC alias to section of copied symbol
commit 8c6a8e12 (2023-12-30): Fix #2469: if expression depends on one symbol and ASMPC...

... do not convert it to the section of the symbol.

This conversion to the section of the single symbol in an expression
was done to fix #2418.
closed 2023-10-20 by pauloscustodio