z88dk/z88dk #2419: [ Feature requrest ] Multithreaded build

atirut-w, 2023-10-16:
Would be something nice to have. Currently, the build script doesn't event saturate one core and it takes a good amount of time to build.

polluks, 2023-10-16:
What about `make -j`?

atirut-w, 2023-10-16:
Most of the time is taken by `./build.sh`, and I'm not quite sure if you can pass args for `make` into it or how to even invoke what `./build.sh` does manually.

I might look into the script and seeing if I can contribute anything here to make it go faster.

suborb, 2023-10-16:
There's a few stages that consume most of the time:

1. Building the libraries 
2. Running the z80asm tests
3. Running the ticks tests

Specifying -z will skip the z80asm tests (2) and the ticks tests are skipped by default (2) unless -t is supplied.

That leaves 1. - there's a lot going on here, the "cheat code" is just to download the nightly tar ball which already has the libraries built (and then specify -l to build.sh to skip building them). 

make -j works seems to work correctly for the binaries.

The (classic) library build has now mostly been changed to a non-recursive make which should track dependencies more accurately which will make rebuilds faster. I've just pushed a resolution to some more -j X issues. Unfortunately building over 300 libraries is never going to be massively fast. 

The (newlib) build is parallelisable.

suborb, 2023-10-22:
We've done quite a lot on this over the past week and it seems that it might be worth other people trying and feeding back - Usually with these things something pops up in a new environment.

Taking classic library build. I've been testing on 3 machines:

On Ubuntu 20 on an i5-9500T:

```
make -j 8

real    2m35.584s
user    12m38.872s
sys     1m19.343s
```

On an M1 Mac mini:

```
make -j 8 353.41s user 523.30s system 598% cpu 2:26.43 total
```

On my laptop (i5-4260U MacOS):

```
make -j  8 1977.01s user 1659.41s system 290% cpu 20:51.80 total
```
 
Bringing this all together and doing `MAKE_CONCURRENCY=-j ./build.sh` on the Linux machine yields:

```
real	3m45.468s
user	17m45.194s
sys	1m38.536s
```

Though doing that will kill the machine (I saw a load average of over 900), so limiting the number of jobs is probably the way to go.

Some of this has been applied to the CI builds and we've seen the Ubuntu time come down from 30+ minutes to average around 23, and the Mac build from 90+ minutes to about an hour.

closed 2023-11-05 by suborb