z88dk/z88dk #2420: (z80asm) Linking fails

polluks, 2023-10-16:
```
Can't locate Path/Tiny.pm in @INC (you may need to install the Path::Tiny module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/x86_64-linux-gnu/perl5/5.28 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at build_ldflags.pl line 4.
BEGIN failed--compilation aborted at build_ldflags.pl line 4.
g++ -o z88dk-z80asm src/c/str.o src/c/libfile.o src/c/z80pass.o src/c/reloc_code.o src/c/parse1.o src/c/list.o src/c/sym.o src/c/dbg.o src/c/directives.o src/c/opcodes.o src/c/common.o src/c/alloc.o src/c/modlink.o src/c/strhash.o src/c/minunit.o src/c/codearea.o src/c/scan1.o src/c/cpu_rules_action.o src/c/zobjfile.o src/c/module1.o src/c/class.o src/c/symtab1.o src/c/expr1.o src/c/array.o src/c/z80asm.o ../common/die.o ../common/strutil.o ../common/z80asm_cpu.o ../common/fileutil.o ../common/objfile.o ../common/zutils.o ../common/option.o ../common/dirname.o ../common/optparse.o ../../ext/regex/regfree.o ../../ext/regex/regcomp.o ../../ext/regex/regexec.o ../../ext/regex/regerror.o src/cpp/errors.o src/cpp/args.o src/cpp/utils.o src/cpp/defines.o src/cpp/preproc.o src/cpp/lstfile.o src/cpp/float.o src/cpp/scan.o src/cpp/main.o

/usr/bin/ld: src/cpp/args.o: in function `std::filesystem::__cxx11::path::path<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::filesystem::__cxx11::path>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::filesystem::__cxx11::path::format) [clone .constprop.156]':
args.cpp:(.text+0x10d3): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: src/cpp/args.o: in function `path_file_exists':
args.cpp:(.text+0x116f): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x1177): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: src/cpp/args.o: in function `path_replace_ext':
args.cpp:(.text+0x1844): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x1873): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x187e): undefined reference to `std::filesystem::__cxx11::path::replace_extension(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: src/cpp/args.o: in function `replace_ext(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
args.cpp:(.text+0x1935): undefined reference to `std::filesystem::__cxx11::path::replace_extension(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: src/cpp/args.o: in function `Args::check_library(std::filesystem::__cxx11::path const&)':
args.cpp:(.text+0x2220): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: src/cpp/args.o: in function `Args::bin_filename(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
args.cpp:(.text+0x2829): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x2870): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x291a): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x2a72): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x2bcd): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: src/cpp/args.o:args.cpp:(.text+0x2cd3): more undefined references to `std::filesystem::__cxx11::path::_M_split_cmpts()' follow
/usr/bin/ld: src/cpp/args.o: in function `Args::bin_filename(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
args.cpp:(.text+0x2d60): undefined reference to `std::filesystem::__cxx11::path::parent_path() const'
/usr/bin/ld: args.cpp:(.text+0x2e4a): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x2e5a): undefined reference to `std::filesystem::__cxx11::path::_M_find_extension() const'
/usr/bin/ld: args.cpp:(.text+0x2efc): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x2f18): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x2f59): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x30a9): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: src/cpp/args.o: in function `Args::check_source(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)':
args.cpp:(.text+0x34e7): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: src/cpp/args.o:args.cpp:(.text+0x35cd): more undefined references to `std::filesystem::__cxx11::path::_M_split_cmpts()' follow
/usr/bin/ld: src/cpp/args.o: in function `Args::check_source(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)':
args.cpp:(.text+0x3604): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: args.cpp:(.text+0x3619): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: args.cpp:(.text+0x37d9): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x38c1): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x395c): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: args.cpp:(.text+0x3ac6): undefined reference to `std::filesystem::last_write_time(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: args.cpp:(.text+0x3ad1): undefined reference to `std::filesystem::last_write_time(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: src/cpp/args.o: in function `Args::search_path(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) [clone .constprop.153]':
args.cpp:(.text+0x3dca): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: args.cpp:(.text+0x3e44): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x3e57): undefined reference to `std::filesystem::__cxx11::path::has_root_directory() const'
/usr/bin/ld: args.cpp:(.text+0x3ef6): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: args.cpp:(.text+0x3f64): undefined reference to `std::filesystem::__cxx11::path::has_filename() const'
/usr/bin/ld: args.cpp:(.text+0x3f8c): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: src/cpp/args.o: in function `Args::search_path(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
args.cpp:(.text+0x4739): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: args.cpp:(.text+0x4798): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: args.cpp:(.text+0x47fd): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: args.cpp:(.text+0x4810): undefined reference to `std::filesystem::__cxx11::path::has_root_directory() const'
[... cut ...]
/usr/bin/ld: preproc.cpp:(.text+0x8f6d): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
/usr/bin/ld: src/cpp/scan.o: in function `FileScanner::open(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
scan.cpp:(.text+0x20f6): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
/usr/bin/ld: scan.cpp:(.text+0x20fe): undefined reference to `std::filesystem::status(std::filesystem::__cxx11::path const&)'
collect2: error: ld returned 1 exit status
make: *** [Makefile:57: z88dk-z80asm] Error 1

suborb, 2023-10-16:
Coincidentally,  I was thinking about this problem last night - I hit it on one of my machines. `-lstdc++fs` needs to be added to the linker:

```
--- a/src/z80asm/Makefile
+++ b/src/z80asm/Makefile
@@ -54,7 +54,7 @@ DEPENDS               := $(C_SRCS:.c=.d) $(CXX_SRCS:.cpp=.d)
 all: $(PROJ)$(EXESUFFIX)

 $(PROJ)$(EXESUFFIX): ../config.h $(OBJS)
-       $(CXX) -o $(PROJ)$(EXESUFFIX) $(OBJS) $(LDFLAGS)
+       $(CXX) -o $(PROJ)$(EXESUFFIX) $(OBJS) $(LDFLAGS) -lstdc++fs

 ../config.h:
        @echo \#define PREFIX \"/usr/local/share/z88dk\"                                 > $@
```

It's not autodetected because I've not quite figured out the right way to do it.

pauloscustodio, 2023-10-16:
Make build_ldflags.pl independent of third party libraries

pauloscustodio, 2023-10-16:
I have fixed ```Can't locate Path/Tiny.pm in @INC``` in build_ldflags.pl

pauloscustodio, 2023-10-17:
Closing this as complete. @polluks, please reopen if the solution does not work on your environment.

closed 2023-10-17 by pauloscustodio