z88dk/z88dk #2423: ez80 - ld.lis opcodes

feilipu, 2023-10-17:
There is an option to load a 2 bytes into an ez80 24 bit register using the `ld.lis` opcode.
I don't think these have been implemented.

For reference, the `BCU`, `DEU`, `HLU` registers are loaded with 0x00, while the normal 16 bit registers are loaded with the following two bytes.

feilipu, 2023-10-17:
Hmm. Not sure now... in um077.pdf on Page 230 the opcode table says it is not possible.
Yet on Page 20 the below example is given.

```
LD.LIS HL, 3456h ;HL[23:0] ¨ {00h, 3456h}. .IS
                 ;directs eZ80 to fetch only 16-
                 ;bits of data. .L uses all 3 bytes
                 ;of HL register.
```
I don't have an Agon to test, I'm afraid.

TheByteAttic, 2023-10-17:
You should ask L Break Into Program (Dean), the firmware designer.

feilipu, 2023-10-17:
Dean @breakintoprogram do you know the answer to this question?

feilipu, 2023-10-17:
OK, so Dean has answered on X and it seems that `ld.lis rr,mm` opcodes are real.
https://twitter.com/BreakIntoProg/status/1714272497120493677