z88dk/z88dk #2426: ZCC config file is hardcoded to `/usr/local/share/z88dk//lib/config/zx.cfg`

atirut-w, 2023-10-17:
This renders using `PREFIX=...` pointless and packaging the SDK into `.deb`, `.rpm`, etc. impossible unless the build script applies a patch.

suborb, 2023-10-17:
There's a rather long list of places where zcc looks for configuration files to cover almost every contingency...

1. In the `ZCCCFG` directory
2. In the current directory
3. In {install_dir}/share/z88dk/lib/config
4. Relative to zcc (if invoked with a path)

The following shows cases 2-4:

```
dom@ermintrude z88dk % unset ZCCCFG
dom@ermintrude z88dk % rm src/config.h bin/zcc
dom@ermintrude z88dk % export PREFIX=/opt/z88dk
dom@ermintrude z88dk % make bin/zcc
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C src/zcc PREFIX=`pwd` install
cc -c -o zcc.o -O2 -DLOCAL_REGEXP -Wall -pedantic -g -MMD -I. -I../copt -I../common -I../../ext/uthash/src zcc.c
zcc.c:1951:5: warning: unused function 'copy_defc_file' [-Wunused-function]
int copy_defc_file(char *name1, char *ext1, char *name2, char *ext2)
    ^
1 warning generated.
cc -o zcc -O2 -DLOCAL_REGEXP -Wall -pedantic -g -MMD zcc.o ../copt/regex/regcomp.o  ../copt/regex/regerror.o ../copt/regex/regexec.o  ../copt/regex/regfree.o ../common/dirname.o ../common/option.o
install zcc /Users/dom/z88dk/bin/
dom@ermintrude z88dk % zcc +unknown
Can't open config file /opt/z88dk/share/z88dk//lib/config/unknown.cfg
dom@ermintrude z88dk % bin/zcc +unknown
Can't open config file bin/../lib/config//unknown.cfg
dom@ermintrude z88dk % touch local.cfg
dom@ermintrude z88dk % zcc +local.cfg
No CRT0 defined in configuration file <local.cfg>
```

It's true that it defaults to `/usr/local/bin` in the top level Makefile, but it can be overridden trivially as above (and that also works via build.sh)

atirut-w, 2023-10-18:
> The following shows cases 2-4:
>
> ```
> dom@ermintrude z88dk % unset ZCCCFG
> dom@ermintrude z88dk % rm src/config.h bin/zcc
> dom@ermintrude z88dk % export PREFIX=/opt/z88dk
> dom@ermintrude z88dk % make [...]

Setting prefix before running `./build.sh` works for me. This should definitely be documented.

suborb, 2023-10-18:
Great, I've updated the installation page with a bit of blurb about it.

closed 2023-10-18 by suborb