z88dk/z88dk #2427: [zcc][unix only][macro?] Name conflict or broken macro preprocessor *only* under Unix

Fabrizio-Caruso, 2023-10-17:
If I compile z88dk from source (very latest source) *under Linux* I get this weird error when I try to build my latest Cross-Lib game

SCCZ80:
```
 games/stinger/main.c:715:8: error: Missing token, expecting ( got ;
games/stinger/main.c:715:8: error: Illegal Argument Name: 
games/stinger/main.c:716:1: error: Illegal Argument Name: 
games/stinger/main.c:717:15: error: Illegal Argument Name: 
games/stinger/main.c:717:18: error: Illegal Argument Name: 
games/stinger/main.c:717:24: error: Illegal Argument Name: 
games/stinger/main.c:718:18: error: Illegal Argument Name: 
games/stinger/main.c:718:20: error: Illegal Argument Name: 
games/stinger/main.c:718:25: error: Illegal Argument Name: 
games/stinger/main.c:718:30: error: Definition of 'lives': int lives(int , int , int next_threshold, int , int , int , int bit_beep_callee, int , int , int ) - does not match declaration at "games/stinger/main.c":262 : unsigned char lives

Maximum (10) number of errors reached, aborting!
Compilation aborted
make: *** [Makefile_common:1701: spectrum_48k] Error 1
```

ZSDCC:
```
games/stinger/main.c:657: warning 84: 'auto' variable 'color' may be used before initialization
games/stinger/main.c:714: syntax error: token -> '{' ; column 13
make: *** [Makefile_common:1701: spectrum_48k] Error 1
```

My code only defines a variable `lives` and not a variable but it seems that Z88DK is also providing 
```int lives(int , int , int next_threshold, int , int , int , int bit_beep_callee, int , int , int )```

I am not getting this issue with the very latest Windows snapshot.
Could a Unix-only macro change the definition of some bit-banging routine only under Unix?

I will try to produce a minimal case...
What else can I do to investigate this?

Fabrizio-Caruso, 2023-10-18:
@zx70 it looks like bit banging is causing name conflicts in the Unix build. Are you aware of the differences between macros under Unix and under Windows?

suborb, 2023-10-18:
I can't find any reference to lives in the code base so I think that's a bogus error message.

So I'm wondering if there's something wrong with the `handle_extra_life()` macro - if you add a -E to the flags it'll stop after preprocessing and we can see what's happening from the .I file generated.

Fabrizio-Caruso, 2023-10-18:
By the way running dos2unix "fixes" the problem but I think this is still a bug because it should work with Windows, MacOS and Linux CR/LF conventions.

suborb, 2023-10-18:
Normally it does handle the line feeds sensibly, but \ macros are possibly a special case:

```
                if (c == '\r') {
                        /*
                         * We found a '\r'; we handle it as a newline.
                         * For '\r\n' cases (Windows) prefer to ignore this one and use '\n' later.
                         */
```

Fabrizio-Caruso, 2023-10-18:
@pauloscustodio thanks. I will.
On the other hand other people may be only using Windows and not git. So I think we have a bug anyway.

Fabrizio-Caruso, 2024-05-09:
I have re-encountered the same problem. The macro preprocessor is still broken in some cases with text using Windows line terminations.

suborb, 2024-05-09:
It's a funky one. The problem occurs when the \r is at the end of the lexer buffer and we can't check for a following \n.

Which explains why it's a bit intermittent and creating a small test case is tricky.

suborb, 2024-05-10:
It's obviously the end of a very long week - I was sure I'd written a reply here!

The preprocessor is about the only bit of the toolchain that we've not really had a hand in - historically we've just used a 3rd party one and imported it, only adding some minor features around the edge and not poking at it too hard.

As it happens, it does look like this was introduced by 73e1e9e871535ae032c23f1a18d7d1b424f6925c whilst attempting to fix a line numbering issue though I am at the moment (see first line), not certain why that change worked.

Regardless, it's actually pretty trivial to change the preprocessor that's used - if you take a look at lib/config/tools.inc you'll see the following:

```
# The most common tool to change is the preprocessor. Uncomment the following
# lines to switch to using the clang preprocessor.

# CPP clang -E -U__clang__
# SDCPP clang -E -U__clang__
# STYLECPP 4
```

Just uncomment them and you can use the system clang preprocessor.

Fabrizio-Caruso, 2024-05-10:
Thanks for the fix. Maybe we might have a parameter to define the preprocessor.

closed 2024-05-10 by Fabrizio-Caruso