z88dk/z88dk #2430: [camputers_lynx][sound][feature] Sound is missing

Fabrizio-Caruso, 2023-10-25:
Hi!
I don't find any sound support in `features.h` for the Camputers Lynx.

In Cross-Lib I may implement it my self by hitting the 6-bit DAC at `$0084` (keep MSB zero and just hit the lowest 6 bits).
On the other hand it would be nice to have one bit audio in Z88DK. It could be implemented by hitting (at least) bit 5 (highest DAC bit).
Noise could be produced by writing random stuff on lowest bits. 

More info here:
https://www.chibiakumas.com/z80/CamputersLynx.php

[image]

suborb, 2023-10-25:
I think that's an omission in features.h  - it looks like there is one bit sound available.

Fabrizio-Caruso, 2023-10-25:
@suborb I am going to recheck it. I could not hear any sound on the emulator. I may have used to wrong settings.

suborb, 2023-10-25:
For reference, the library config looks right:

```
        defc SOUND_ONEBIT_port   = $84
        defc SOUND_ONEBIT_bit    = 5
        defc SOUND_ONEBIT_mask   = 32
```

Fabrizio-Caruso, 2023-10-25:
@suborb if it writes the mask value 32 = (00100000), then we are safe, it could even write 63 to produce a sound with almost double volume.

Fabrizio-Caruso, 2023-10-25:
Sound works under Jinx emulator. So it should be OK.

Fabrizio-Caruso, 2023-10-25:
I think the only error is that there is no mention in features.h unless I overlooked it.

suborb, 2023-10-26:
Yup, I fixed that in [7d58db7](https://github.com/z88dk/z88dk/commit/7d58db7c8ba35b0ecc00b1e71f41fec3390d0ed6)

closed 2023-10-26 by suborb