z88dk/z88dk #2432: [pc88] Wrong colors (green->red, red->green, cyan->purple)

Fabrizio-Caruso, 2023-10-27:
Hi

On the PC88 some common colors are very wrong.

Some wrong colors
- red and green are switched.
- cyan is displayed as purple.

Some correct colors
- yellow 
- white 
- black 

WRONG (PC88):
[image]

CORRECT (MSX):
[image]

suborb, 2023-10-31:
Sorry, I've been on other things and forgot about this one.

On machines with palette registers I've tended not to bother mapping and rely on the palette being set.

On quite a lot of machines I've manually set the palette on startup.

Unfortunately on the pc88 I didn't - the registers are defined in `<arch/pc88.h>` so as a workaround you could set the palette yourself:

```
__sfr __at 0x54 PAL_REG0;       // bits 0-2 contain  palette value 0-7
__sfr __at 0x55 PAL_REG1;       // bits 0-2 contain  palette value 0-7
__sfr __at 0x56 PAL_REG2;       // bits 0-2 contain  palette value 0-7
__sfr __at 0x57 PAL_REG3;       // bits 0-2 contain  palette value 0-7
__sfr __at 0x58 PAL_REG4;       // bits 0-2 contain  palette value 0-7
__sfr __at 0x59 PAL_REG5;       // bits 0-2 contain  palette value 0-7
__sfr __at 0x5a PAL_REG6;       // bits 0-2 contain  palette value 0-7
__sfr __at 0x5b PAL_REG7;       // bits 0-2 contain  palette value 0-7
```

Fabrizio-Caruso, 2023-10-31:
By poking at those addresses? Each register is one palette color with 8 possible values? (only 3 bit of color resolution in all screen modes?)

suborb, 2023-10-31:
Yes:

```
PAL_REG0 = 0...7
```

There's only 8 colours (including black)

Fabrizio-Caruso, 2023-11-01:
I also need to know which color macro is mapped to which index... Well I can figure it out by trial and error....

Can I set all other bits to zero?

Fabrizio-Caruso, 2023-11-01:
@suborb "poking" does not work.
I fear I cannot do what you suggest in C. Maybe we need Assembly OUT commands...

The following code (even with different values on the registers) has no effect on the colors.
```
	#if defined(__PC88__)
		#define COLOR_REG 0x54
		#define POKE(addr,val)     (*(uint8_t*) (addr) = (val))

		POKE(COLOR_REG+3,3);
		POKE(COLOR_REG+2,2);
		POKE(COLOR_REG+4,4);
		/*
		{
			uint8_t i;
			_XL_PRINTD(0,0,2,_XL_BLACK); // 0
			_XL_PRINTD(0,1,2,_XL_CYAN);  // 3
			_XL_PRINTD(0,2,2,_XL_GREEN); // 2
			_XL_PRINTD(0,3,2,_XL_RED);   // 4
			_XL_PRINTD(0,4,2,_XL_WHITE); // 15
			_XL_PRINTD(0,5,2,_XL_YELLOW); // 14
			
			while(1)
			{
			}
		}
		*/
	#endif
```

Fabrizio-Caruso, 2023-11-01:
I was right about POKE not working...

This has some effect (yet, not the right because of wrong values):
```
		#asm
			LD A,3
			OUT ($54+3),A
			
			LD A,2
			OUT ($54+2),A
			
			LD A,4
			OUT ($54+4),A
		#endasm
```

suborb, 2023-11-01:
Your C syntax was incorrect, those variables are sfr's so you just do this (eg to set pen7 to yellow):

```
    PAL_REG7 = 6;
```

Regardless I've added in proper mapping now.

Fabrizio-Caruso, 2023-11-01:
Thanks! Sorry for the wrong syntax. I was not aware of the sfr syntax.

closed 2023-11-05 by suborb