z88dk/z88dk #2440: Memory corruption with switch-local buffer

dfandrich, 2023-11-09:
A buffer placed in the scope of a switch statement results in memory corruption. The following code:
```c
#include <stdio.h>
int main(void) {
   /* char switchbuf[32]; // Placing the buffer here works fine */
  int a=1234;
  printf("a: %d\n", a);
  switch (a) {
    char switchbuf[32];
    case 1234:
      printf("a: %d\n", a);
      break;
  }
  printf("a: %d\n", a);
  return 0;
}
```
results in the output:
```
a: 1234
a: 0
a: 0
```
when it should result in:
```
a: 1234
a: 1234
a: 1234
```
Moving the `switchbuf` variable outside the switch statement works as expected.

This is using the code from git from about 2 months ago.

dfandrich, 2023-11-10:
The test program was compiled with `zcc +cpm -create-app switchvar.c`  I've confirmed that the problem still exists with at HEAD 162135bd.

suborb, 2023-11-10:
Impressive edge case - I've not come across code that does that.

It may take me a while to figure out how to solve this one nicely.

dfandrich, 2023-11-10:
Speaking of edge cases, I should see if zcc supports Duff's device ☺

suborb, 2023-11-10:
I believe it does!

closed 2026-10-27 by feilipu