z88dk/z88dk #2447: [z9001] UDG KRT mode may be broken

Fabrizio-Caruso, 2023-11-22:
I have tried to build my Cross-Lib games for the Z9001 in mode 1, for which Z88DK is supposed to support UDG for KRT graphics.

Under JKCEMU emulator, characters are not redefined. Maybe I am doing something wrong. Maybe I am not setting the emulator correctly. 
This is what I get:
[image]

This is what I should see:
[image]

My gfx settings:
[image]

Before I write a minimal example, could any of you confirm UDGs are supported and I only need to set the mode to 1 and nothing else?

suborb, 2023-11-22:
That should be all that’s needed. 

The screenshot looks like the mode isn’t being changed at all!

I‘ll test in a bit.

suborb, 2023-11-22:
My little test program shows this:

[image]

with jkcemu 0.9.7

Fabrizio-Caruso, 2023-11-22:
@suborb, Sorry, is your example showing user-defined tiles?
What do I need to do to switch the mode?
The doc says I should use mode 1.
I do:
```
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
	int mode = __SCREEN_MODE;
	console_ioctl(IOCTL_GENCON_SET_MODE, &mode); 
```
where `__SCREEN_MODE` is 1. I have also tried 2.

My Makefile target is:
```
z9001: $(ASSETS_PATH)/z88dk_xchase.asm
	$(Z88DK_PATH)$(MYZ88DK) +z9001 $(SCCZ80_OPTS) $(INCLUDE_OPTS) -clib=ansi \
	-D__Z9001__ -vn  -D__SCREEN_MODE=1 -D_XL_NO_JOYSTICK  \
	-D__CONIO_GRAPHICS  -D__BIT_BANG_SOUND \
	-lndos -create-app -o $(BUILD_PATH)/X$(GAME_NAME)_$@.z80 \
	$(FULL_FILES) \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(ASSETS_PATH)/z88dk_xchase.asm
```

suborb, 2023-11-22:
So in my screenshot you can see it's using the ZX font and there's 3/4 of a smiling face.

In your screenshot it looks the the Z9001 default font is being used which suggests that the screen mode isn't being changed.

To try and figure out where it's going wrong, I've run a test where I don't change the emulated hardware (so left on "S/W Blockgrafik") but do switch mode and I end up with a blank screen.

So, the fact that you've got colour means that you've correctly switched the emulated hardware, but somehow this bit of code:

```
	void *param = &udgs;
	console_ioctl(IOCTL_GENCON_SET_FONT32, &param);
	int mode = __SCREEN_MODE;
	console_ioctl(IOCTL_GENCON_SET_MODE, &mode);
```

isn't being executed or taking effect.

Looking closer at your makefile:

```
z9001: $(ASSETS_PATH)/z88dk_xchase.asm
	$(Z88DK_PATH)$(MYZ88DK) +z9001 $(SCCZ80_OPTS) $(INCLUDE_OPTS) -clib=ansi \
```

Remove -clib=ansi and replace with --generic-console As is usual, the VT100 terminal only targets one screen mode, and for the z9001 it's the text screen mode.

Fabrizio-Caruso, 2023-11-22:
Thanks for your replay.

If I replay -clib=ansi with --generic-console, nothing changes.
On the other hand, if I just remove -clib=ansi, I get some gibberish (with graphics) on the screen. 

I will need to see how I can reproduce this with a smaller example...

Within Cross-Lib I just do `xl bomber z9001` to compile the smallest of my games

[image]

I have attached the binaries.
[Xbomber_z9001.zip](https://github.com/z88dk/z88dk/files/13444662/Xbomber_z9001.zip)

suborb, 2023-11-22:
> Within Cross-Lib I just do xl bomber z9001 to compile the smallest of my games

Unfortunately that fails with the following for me (I think I've had this trouble in the past):

```
Traceback (most recent call last):
  File "./generate_assets.py", line 284, in <module>
    main()
  File "./generate_assets.py", line 279, in main
    generate_assets_from_dir(dir_name)
  File "./generate_assets.py", line 270, in generate_assets_from_dir
    generate_asset_from_template(dir_name, stripped_template_file_name)
  File "./generate_assets.py", line 263, in generate_asset_from_template
    process_a7800_160A_file()
  File "./generate_assets.py", line 202, in process_a7800_160A_file
    print("Asset has wrong length: " << len(source))
TypeError: unsupported operand type(s) for <<: 'str' and 'int'
make: *** [games/bomber/generated_assets/z88dk_xchase.asm] Error 1
```

> I have attached the binaries.

Can I have the .map as well?

Fabrizio-Caruso, 2023-11-23:
@suborb Sorry for the stupid Python error.

I have fix it now. If you pull the latest version, it should now work for you.

I have attached the map files and the binaries produced by these Makefile targets:
```
z9001: $(ASSETS_PATH)/z88dk_xchase.asm
	$(Z88DK_PATH)$(MYZ88DK) +z9001  -m  $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-D__Z9001__ -vn  -D__SCREEN_MODE=1+32 -D_XL_NO_JOYSTICK  \
	-D__CONIO_GRAPHICS  -D__BIT_BANG_SOUND \
	-lndos -create-app -o $(BUILD_PATH)/X$(GAME_NAME)_$@.z80 \
	$(FULL_FILES) \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(ASSETS_PATH)/z88dk_xchase.asm

z9001_generic_console: $(ASSETS_PATH)/z88dk_xchase.asm
	$(Z88DK_PATH)$(MYZ88DK) +z9001  -m  $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
	-D__Z9001__ -vn  -D__SCREEN_MODE=1+32 -D_XL_NO_JOYSTICK  \
	-D__CONIO_GRAPHICS  -D__BIT_BANG_SOUND \
	-lndos -create-app -o $(BUILD_PATH)/X$(GAME_NAME)_$@.z80 \
	$(FULL_FILES) \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(ASSETS_PATH)/z88dk_xchase.asm
```

[build.zip](https://github.com/z88dk/z88dk/files/13447346/build.zip)

suborb, 2023-11-23:
Thanks - I can now build and play.

```
--- a/src/Makefile_common
+++ b/src/Makefile_common
@@ -1919,7 +1919,7 @@ z9001_broken_gfx: $(ASSETS_PATH)/z88dk_xchase.asm
 # --generic-console
 z9001: $(ASSETS_PATH)/z88dk_xchase.asm
        $(Z88DK_PATH)$(MYZ88DK) +z9001  -m  $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
-       -D__Z9001__ -vn  -D__SCREEN_MODE=1+32 -D_XL_NO_JOYSTICK  \
+       -D__Z9001__ -vn  -D__SCREEN_MODE=1 -D_XL_NO_JOYSTICK --generic-console  \
        -D__CONIO_GRAPHICS  -D__BIT_BANG_SOUND \
        -lndos -create-app -o $(BUILD_PATH)/X$(GAME_NAME)_$@.z80 \
        $(FULL_FILES) \
@@ -1928,7 +1928,7 @@ z9001: $(ASSETS_PATH)/z88dk_xchase.asm

 z9001_generic_console: $(ASSETS_PATH)/z88dk_xchase.asm
        $(Z88DK_PATH)$(MYZ88DK) +z9001  -m  $(SCCZ80_OPTS) $(INCLUDE_OPTS) \
-       -D__Z9001__ -vn  -D__SCREEN_MODE=1+32 -D_XL_NO_JOYSTICK  \
+       -D__Z9001__ -vn  -D__SCREEN_MODE=1 -D_XL_NO_JOYSTICK  \
        -D__CONIO_GRAPHICS  -D__BIT_BANG_SOUND \
        -lndos -create-app -o $(BUILD_PATH)/X$(GAME_NAME)_$@.z80 \
        $(FULL_FILES) \
```

[image]

Fabrizio-Caruso, 2023-11-23:
@suborb ! So, something is wrong in my setup or very special in yours.
Which Makefile target did you use? Which settings did you use on the emulator?

Even with the latest JKCEMU (0.9.8.3), with KRT, I get gibberish on the screen:
[image]

suborb, 2023-11-23:
So, if I run the binaries you attached last of all I can reproduce what you see.

I can see two problems in the binary which I think together are causing the issue:

1. The screen mode is incorrect (you added 32 to it)
2. The screen printer is using the firmware

I'm using the `xl bomber z9001` using my makefile changes above - can you try those and if it's still failing attach the binaries and map files again.

Fabrizio-Caruso, 2023-11-23:
Thanks @suborb!
It works indeed! 
I still don't understand why I need `--generic-console` in this particular target and not in all the others.
My guess is that the standard console for this specific target is vt100/ansi, which is locked to one mode, which does not happen to support udg. Is this the reason?

suborb, 2023-11-23:
Well you are! `--generic-console` is an alias for `-pragma-redirect:fputc_cons=fputc_cons_generic` which you are using for a lot of targets.

Fabrizio-Caruso, 2023-11-23:
@suborb I should switch to the more readable `--generic-console`. 
My bad (as usual). Thanks for your patience and support

closed 2023-11-23 by Fabrizio-Caruso