z88dk/z88dk #2456: (z80asm) Crash on s390 when building library

suborb, 2023-12-06:
From the snap build:

```
--- Building z80 crt Library ---

NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/z80/z80 -I.. -mz80 @crt0_z80.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/ixiy/ixiy -I.. -IXIY -D__SDCC_IX  @crt0_z80.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/8080/8080 -I..  -DCPU_8080 -m8080 @crt0_8080.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/8085/8085 -I..  -DCPU_8085 -m8085 @crt0_8085.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/r2ka/r2ka -I.. -mr2ka  @crt0_r2ka.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/gbz80/gbz80 -I..  -DCPU_GBZ80 -mgbz80 @crt0_gbz80.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/z180/z180 -I..  -mz180 @crt0_z180.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/ez80_z80/ez80 -I..  -mez80_z80 -DEZ80 @crt0_ez80.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/z80n/z80n -I.. -mz80n  @crt0_z80n.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/r4k/r4k -I.. -mr4k  @crt0_r2ka.lst
NEWLIB_ROOT=../_DEVELOPMENT/ z88dk-z80asm -Oobj/kc160/kc160 -I.. -mkc160  @crt0_kc160.lst
TYPE=z80 z88dk-z80asm -d -I/build/z88dk/parts/part-z88dk/build/lib/config//../ -mz80 -DSTANDARDESCAPECHARS -x/build/z88dk/parts/part-z88dk/build/libsrc//z80_crt0 @/build/z88dk/parts/part-z88dk/build/libsrc//../libsrc//z80.lst
Segmentation fault (core dumped)
```

Assembling to object files seems to work.

pauloscustodio, 2023-12-08:
I could not reproduce with spacraft running in Windows WSL, Ubuntu and Darwin. All of them built successfully.

Is it possible to run gdb on the core file and run the backtrace command in gdb, to see where the process crashed?

suborb, 2023-12-08:
Unfortunately we've got no access to the containers used so can't get out any diagnostics. I think we've got two options:

1. Add a SEGV handler which walks the stack (similar to java hs_err* file handling), not that trivial since we're called from a signal handler but maybe `backtrace_symbols_fd([preallocated buffer],[preallocated len], stderr)` is signal safe.
2. Run binaries from a script which enables core files and then runs gstack on the core if it crashes

pauloscustodio, 2023-12-08:
Can you try ```snapcraft --debug```? It should land you in a shell after the SIGSEGV from z80asm.

suborb, 2023-12-08:
Ah, I think I’ve not been clear. 

This is running on the Ubuntu servers - not mine! - otherwise I’d have given you a stack trace already!

pauloscustodio, 2023-12-08:
Commit 5ca8eacefebee383035d50f9b5c56ba87e126607  on branch issue_2456 shows a stack dump and exits on SIGSEGV; it only compiles in Linux.
Can you test with this?

pauloscustodio, 2023-12-11:
@suborb, is it possible to tell the Ubuntu servers to use a special branch, or do we need the stack trace in the main branch? In the later case I would need to add some compilation guards to only do the stack trace on Linux.

suborb, 2023-12-12:
```
TYPE=z80 z88dk-z80asm -d -I/build/z88dk/parts/part-z88dk/build/lib/config//../ -mz80 -DSTANDARDESCAPECHARS -x/build/z88dk/parts/part-z88dk/build/libsrc//z80_crt0 @/build/z88dk/parts/part-z88dk/build/libsrc//../libsrc//z80.lst
Got SIGSEGV=11 at address: 0x2aa535bf000
backtrace() returned 15 addresses:
z88dk-z80asm(dump_stack+0x4a)[0x2aa0eefd042]
z88dk-z80asm(+0x27d0d8)[0x2aa0eefd0d8]
[0x3fff0b7e300]
z88dk-z80asm(strtable_add_string+0x30)[0x2aa0eefe478]
z88dk-z80asm(strtable_parse+0x72)[0x2aa0eeff5e2]
z88dk-z80asm(+0x16af16)[0x2aa0edeaf16]
z88dk-z80asm(+0x16af8c)[0x2aa0edeaf8c]
z88dk-z80asm(+0x16b3ce)[0x2aa0edeb3ce]
z88dk-z80asm(object_file_check_append+0xfc)[0x2aa0edec394]
z88dk-z80asm(assemble_file+0xb2)[0x2aa0ed40712]
z88dk-z80asm(z80asm_main+0xc6)[0x2aa0ed409ce]
z88dk-z80asm(main+0x19e)[0x2aa0ed3fc36]
/lib/s390x-linux-gnu/libc.so.6(__libc_start_main+0x10a)[0x3ff8d92440a]
z88dk-z80asm(+0xbfda4)[0x2aa0ed3fda4]
[0x0]
```

pauloscustodio, 2023-12-13:
Added some asserts on the crashing function, waiting to see which input pointer is invalid.

suborb, 2023-12-13:
None of the assertions are being hit:

``
TYPE=z80 z88dk-z80asm -d -I/build/z88dk/parts/part-z88dk/build/lib/config//../ -mz80 -DSTANDARDESCAPECHARS -x/build/z88dk/parts/part-z88dk/build/libsrc//z80_crt0 @/build/z88dk/parts/part-z88dk/build/libsrc//../libsrc//z80.lst
Got SIGSEGV=11 at address: 0x2aa7e107000
backtrace() returned 15 addresses:
z88dk-z80asm(dump_stack+0x4a)[0x2aa3857d042]
z88dk-z80asm(+0x27d0d8)[0x2aa3857d0d8]
[0x3ffe3cfe3f0]
z88dk-z80asm(strtable_add_string+0x42)[0x2aa3857e48a]
z88dk-z80asm(strtable_parse+0x72)[0x2aa3857f62a]
z88dk-z80asm(+0x16af16)[0x2aa3846af16]
z88dk-z80asm(+0x16af8c)[0x2aa3846af8c]
z88dk-z80asm(+0x16b3ce)[0x2aa3846b3ce]
z88dk-z80asm(object_file_check_append+0xfc)[0x2aa3846c394]
z88dk-z80asm(assemble_file+0xb2)[0x2aa383c0712]
z88dk-z80asm(z80asm_main+0xc6)[0x2aa383c09ce]
z88dk-z80asm(main+0x19e)[0x2aa383bfc36]
/lib/s390x-linux-gnu/libc.so.6(__libc_start_main+0x10a)[0x3ff856a440a]
z88dk-z80asm(+0xbfda4)[0x2aa383bfda4]
[0x0]
make: *** [Makefile:2139: z80_crt0.lib] Error 1
``

suborb, 2023-12-13:
snapcraft seems to be having some infrastructure issues for the past few hours at the moment so I can't grab logs.

I've resent the invite so you should be able to login and look at them yourself.

suborb, 2023-12-14:
Build reports are (finally) working again.

However, I've also managed to reproduce on a qemu vm:

```
(gdb) bt full
#0  strlen () at ../sysdeps/s390/strlen-vx.S:42
No locals.
#1  0x000002aa29d03364 in strtable_add_string (st_=0x2aa2b14bde0, str=0x2aa6f1321e4 <error: Cannot access memory at address 0x2aa6f1321e4>)
    at ../common/strtable.c:82
        _uthash_hfstr_keylen = 706424632
        st = 0x2aa2b14bde0
        found = 0x2aa2b130a84
#2  0x000002aa29d05bb0 in strtable_parse (mem=0x2aa2b1321dc "\021") at ../common/strtable.c:223
        pos = 0
        str = 0x2aa6f1321e4 <error: Cannot access memory at address 0x2aa6f1321e4>
        id = 0
        st = 0x2aa2b14bde0
        little_endian = false
        ptrs = 0x2aa2b1321dc
        num_strings = 285212672
        strings = 0x2aa6f1321e4 <error: Cannot access memory at address 0x2aa6f1321e4>
#3  0x000002aa29c227ee in parse_string_table (obj=0x2aa2b128180) at src/c/modlink.c:145
        save_pos = 436
#4  0x000002aa29c2284e in parse_st_str (obj=0x2aa2b128180) at src/c/modlink.c:152
        id = 722633088
#5  0x000002aa29c23006 in object_file_reserve_space (obj=0x2aa2b128180, module=0x2aa2b130a84) at src/c/modlink.c:253
        modname = 0x2aa2b14f470 "assert/obj/z80/assert.o"
#6  0x000002aa29c2323a in object_file_check_append (filename=0x2aa2b14f470 "assert/obj/z80/assert.o", module=0x2aa2b130a84, reserve_space=true, no_errors=false)
    at src/c/modlink.c:305
        obj = 0x2aa2b128180
#7  0x000002aa29ceddac in assemble_file (filename=<optimized out>) at src/c/z80asm.c:72
        src_filename = <optimized out>
        obj_filename = 0x2aa2b14f470 "assert/obj/z80/assert.o"
        got_obj = <optimized out>
        module = <optimized out>
#8  0x000002aa29cee060 in z80asm_main () at src/c/z80asm.c:142
```

pauloscustodio, 2023-12-14:
Got it! It's not working on big-endian systems, it's the little-to-big endian convertion.
Now let's track the b*****d!
Thanks.

suborb, 2023-12-14:
> Got it! It's not working on big-endian systems, it's the little-to-big endian conversion.
> Now let's track the b*****d!
> Thanks.

For reference, the object file generated is the same as on Intel box (which is good).

pauloscustodio, 2023-12-15:
All builds successful, so this issue is closed.

@suborb, are you notified if it ever fails again?

closed 2023-12-15 by pauloscustodio

suborb, 2023-12-15:
> @suborb, are you notified if it ever fails again?

Sadly not, I just check periodically.

Thanks for sorting out the issues.