z88dk/z88dk #2467: z88dk build script fails with error message on macOS Apple Silicon "ld: warning: ignoring file '/usr/local/Cellar/gmp/6.2.1_1/lib/libgmp.10.dylib': found architecture 'x86_64', required architecture 'arm64'"

dbolli, 2023-12-27:
homebrew on Intel x86_64 installs to /usr/local/Cellar
homebrew on Apple Silicon arm64 installs to /opt/homebrew/Cellar

as per https://mac.install.guide/homebrew/index.html

build.sh script appears to only use /usr/local in the library search path

Dependencies such as [libgmp](https://gmplib.org) cause an error message
when running build script on macOS Apple Silicon.

Suggest that /opt/homebrew/ needs to be pre-pended to /usr/local/ in the
library search path if Apple Silicon is detected.

Workaround is to copy libraries from /opt/homebrew/ to /usr/local and manually
re-create symlinks

```
$ cd /usr/local/Cellar
$ mv gmp gmp-old
$ cp -fvR /opt/homebrew/Cellar/gmp .
```

```
$ cd /usr/local/include
$ rm gmp.h
$ ln -s ../cellar/gmp/6.3.0/include/gmp.h
$ ls -l gmp.h
lrwxr-xr-x  1 dbolli  admin  33 27 Dec 14:09 gmp.h -> ../cellar/gmp/6.3.0/include/gmp.h
$ rm gmpxx.h
$ ln -s ../cellar/gmp/6.3.0/include/gmpxx.h
$ ls -l gmpxx.h
lrwxr-xr-x  1 dbolli  admin  35 27 Dec 16:05 gmpxx.h -> ../cellar/gmp/6.3.0/include/gmpxx.h
```

```
$ ls -l /usr/local/lib/libgmp*
lrwxr-xr-x  1 501  admin  41 21 Oct  2021 /usr/local/lib/libgmp.10.dylib -> ../Cellar/gmp/6.2.1_1/lib/libgmp.10.dylib
lrwxr-xr-x  1 501  admin  34 21 Oct  2021 /usr/local/lib/libgmp.a -> ../Cellar/gmp/6.2.1_1/lib/libgmp.a
lrwxr-xr-x  1 501  admin  38 21 Oct  2021 /usr/local/lib/libgmp.dylib -> ../Cellar/gmp/6.2.1_1/lib/libgmp.dylib
lrwxr-xr-x  1 501  admin  42 21 Oct  2021 /usr/local/lib/libgmpxx.4.dylib -> ../Cellar/gmp/6.2.1_1/lib/libgmpxx.4.dylib
lrwxr-xr-x  1 501  admin  36 21 Oct  2021 /usr/local/lib/libgmpxx.a -> ../Cellar/gmp/6.2.1_1/lib/libgmpxx.a
lrwxr-xr-x  1 501  admin  40 21 Oct  2021 /usr/local/lib/libgmpxx.dylib -> ../Cellar/gmp/6.2.1_1/lib/libgmpxx.dylib
$ cd ../lib
$ ls -l ../Cellar/gmp/6.3.0/lib/libgmp.10.dylib
-r--r--r--  1 dbolli  admin  452816 27 Dec 12:42 ../Cellar/gmp/6.3.0/lib/libgmp.10.dylib
$ rm /usr/local/lib/libgmp.10.dylib
$ ln -s ../Cellar/gmp/6.3.0/lib/libgmp.10.dylib
$ rm /usr/local/lib/libgmp.a
$ ln -s ../Cellar/gmp/6.3.0/lib/libgmp.a
$ rm /usr/local/lib/libgmp.dylib
$ ln -s ../Cellar/gmp/6.3.0/lib/libgmp.dylib
$ rm /usr/local/lib/libgmpxx.4.dylib
$ ln -s ../Cellar/gmp/6.3.0/lib/libgmpxx.4.dylib
$ rm /usr/local/lib/libgmpxx.a
$ ln -s ../Cellar/gmp/6.3.0/lib/libgmpxx.a
$ rm /usr/local/lib/libgmpxx.dylib
$ ln -s ../Cellar/gmp/6.3.0/lib/libgmpxx.dylib
$ ls -l /usr/local/lib/libgmp*
lrwxr-xr-x  1 dbolli  admin  39 27 Dec 14:14 /usr/local/lib/libgmp.10.dylib -> ../Cellar/gmp/6.3.0/lib/libgmp.10.dylib
lrwxr-xr-x  1 dbolli  admin  32 27 Dec 14:15 /usr/local/lib/libgmp.a -> ../Cellar/gmp/6.3.0/lib/libgmp.a
lrwxr-xr-x  1 dbolli  admin  36 27 Dec 14:15 /usr/local/lib/libgmp.dylib -> ../Cellar/gmp/6.3.0/lib/libgmp.dylib
lrwxr-xr-x  1 dbolli  admin  40 27 Dec 14:16 /usr/local/lib/libgmpxx.4.dylib -> ../Cellar/gmp/6.3.0/lib/libgmpxx.4.dylib
lrwxr-xr-x  1 dbolli  admin  34 27 Dec 14:16 /usr/local/lib/libgmpxx.a -> ../Cellar/gmp/6.3.0/lib/libgmpxx.a
lrwxr-xr-x  1 dbolli  admin  38 27 Dec 14:16 /usr/local/lib/libgmpxx.dylib -> ../Cellar/gmp/6.3.0/lib/libgmpxx.dylib
```

```
$ cd /usr/local/opt
$ rm /usr/local/opt/gmp
$ ln -s ../Cellar/gmp/6.3.0 gmp
$ ls -l gmp
lrwxr-xr-x  1 dbolli  admin  19 27 Dec 15:58 gmp -> ../Cellar/gmp/6.3.0
$ cd /usr/local/var/homebrew/linked
$ rm gmp
$ ln -s ../../../Cellar/gmp/6.3.0 gmp
$ ls -l gmp
lrwxr-xr-x  1 dbolli  admin  25 27 Dec 16:29 gmp -> ../../../Cellar/gmp/6.3.0
```

```
$ cd /usr/local/Cellar
$ rm -rf gmp-old

$ sudo find /usr/local -name "*gmp*" -ls
```

And fix any remaining symlinks as per above

suborb, 2023-12-27:
Oops, I did think I'd sorted this one out, but I've just tried it out and it appears not.

You should be able to do this prior to running build.sh:

```
export CFLAGS="`pkg-config gmp --cflags` -g -O2"
export LDFLAGS="`pkg-config gmp --libs-only-L` -g -O2"
```

dbolli, 2023-12-27:
Build script still appears to be looking in /usr/local and fails with error message

```
ti8xk.c:20:10: fatal error: 'gmp.h' file not found
#include <gmp.h> // Not much you can do about The GNU multi-percision library 
         ^~~~~~~
1 error generated.
```

suborb, 2023-12-27:
So maybe pig-config isn't working for you? Here's my output:

```
% export CFLAGS="`pkg-config gmp --cflags` -g -O2"
% export LDFLAGS="`pkg-config gmp --libs-only-L` -g -O2"
% env | egrep "CFLAGS|LDFLAGS"
CFLAGS=-I/opt/homebrew/Cellar/gmp/6.3.0/include -g -O2
LDFLAGS=-L/opt/homebrew/Cellar/gmp/6.3.0/lib -g -O2
% make -C src/appmake clean && rm -f bin/z88dk-appmake
% ./build.sh
...
gcc -I/opt/homebrew/Cellar/gmp/6.3.0/include -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common   -c -o ti8xk.o ti8xk.c
...
gcc -I/opt/homebrew/Cellar/gmp/6.3.0/include -g -O2 -Wall -std=gnu11 -MMD -pedantic -I../common -o z88dk-appmake ...
```

dbolli, 2023-12-28:
No, it works but I had to add [-E](https://stackoverflow.com/a/50603486) as in

```
sudo -E ./build.sh
``` 

to preserve the environment variables

Thanks for the workaround 👍

suborb, 2024-03-11:
This is a bit of a universal problem, fwiw I'm now got this in my .profile following the call to `brew shellenv`

```
export CPATH="$HOMEBREW_PREFIX/include:$CPATH"
export LIBRARY_PATH="$HOMEBREW_PREFIX/lib:$LIBRARY_PATH"
```

dbolli, 2024-03-13:
Thanks for the update. Works for me 👍

Nightly build for 20240312 compiled successfully on macOS 14.4 after commenting out
the "export CFLAGS" and "export LDFLAGS" commands listed above and adding the CPATH
and LIBRARY_PATH commands.

Great work 🥰

suborb, 2024-03-13:
Fab. that's now in the [installation page](https://github.com/z88dk/z88dk/wiki/installation#building-on-macos) so I'll close this issue.

closed 2024-03-13 by suborb