z88dk/z88dk #2469: z80asm using $ from incorrect section when computing using defc constant

aralbrec, 2023-12-30:
This one was a little tricky to track down.  Here is a code snippet that produces the problem but must be in a larger context to build incorrectly.

```
SECTION PAGE_93_EnemyCode3
                            include "./Boulder2Roll.asm"
defc SprData_StartPage = 50    ;;;;; AAAAAAAAAAAAAAAAAAAAAAAA
SECTION PAGE_50_spriteData_Enemies3
;defc SprData_StartPage = 50     ;;;;; BBBBBBBBBBBBBBBBBBBBBBBB
public _Boulder2PalettePage

public _Boulder2PalettePage2
public _Boulder2PalettePage3
defc SprData_StartPage2 = 50
defs $4e00
defc _Boulder2PalettePage = SprData_StartPage+($ / $2000)   ;;;; CCCCCCCC
defc _Boulder2PalettePage2 = SprData_StartPage2+($ / $2000)
defc _Boulder2PalettePage3 = 50+($ / $2000)

public _Boulder2Palette
_Boulder2Palette:		defs 1000
_Boulder2Roll_DataPtr:	defs 2000
```

In this snippet, `_Boulder2PalettePage`, `_Boulder2PalettePage2` and `_Boulder2PalettePage3` are expected to have the same value.  However, it was found that `_Boulder2PalettePage` is different.  The correct value is generated if the `SprData_StartPage" defc in position AAAA above is moved to position BBBBB in a different section.

What seems to be happening is the calculation at CCCCC is using `$` from the section `SorData_StartPage` is defined in rather than the current section `$` appears in.  This is a change from a previous version of z88dk-z80asm that was being used before a recent sync with the repo.

pauloscustodio, 2023-12-30:
Hi @aralbrec , nice to hear from you, and thanks for reporting.

The code snippet you sent does not show the error. I've committed it to branch issue_2469, and all ```_Boulder2paletePageN``` have the same value. 

Cany you provide a code that actually fails?
I closed an issue in the same area, #2418, maybe they are related?

aralbrec, 2023-12-30:
Hi @pauloscustodio, 

I'll try to attach a zip here.  The sections need to have different amounts of data in them for the $ to return different section offsets.

The file `SpriteData_Boulder2.asm` contains the error.  When the declaration of `SprData_StartPage` is moved from position AAAA to BBBB, the result is correct. 

As is, the incorrect build produces this in the map file:

```
_Boulder2PalettePage            = $0034 ; addr, public, , SpriteData_Boulder2_asm, PAGE_93_EnemyCode3, SpriteData_Boulder2.asm:12
_Boulder2PalettePage2           = $0037 ; addr, public, , SpriteData_Boulder2_asm, PAGE_50_spriteData_Enemies3, SpriteData_Boulder2.asm:13
_Boulder2PalettePage3           = $0037 ; addr, public, , SpriteData_Boulder2_asm, PAGE_50_spriteData_Enemies3, SpriteData_Boulder2.asm:14
```
when all three constants should be $37.

Build with:

```
zcc +zxn --no-crt -m -v SpriteData_Boulder.asm SpriteData_Boulder2.asm -o sb.bin
```
[issue_2469.zip](https://github.com/z88dk/z88dk/files/13798464/issue_2469.zip)

commit 8c6a8e12 (2023-12-30): Fix #2469: if expression depends on one symbol and ASMPC...

... do not convert it to the section of the symbol.

This conversion to the section of the single symbol in an expression
was done to fix #2418.
closed 2023-12-30 by pauloscustodio