z88dk/z88dk #2473: (help wanted-macosx) m4 error in macosx

pauloscustodio, 2023-12-30:
In one branch I have the following file to be processed by m4:
```
define(`x',INPUT)x
```
and calling ```m4 -DINPUT=nop  "f1.asm.m4"``` I expect the result ```nop```.

On the macosx workflow it is failing:
```
/Applications/Xcode_14.2.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/m4: INTERNAL ERROR: recursive push_string!
sh: line 1: 38786 Abort trap: 6           m4 -DINPUT=nop "f1.asm.m4" > "f1.asm"
m4 -DINPUT=nop  "f1.asm.m4" > "f1.asm": No such file or directory
```

It is running OK on all other workflows.

suborb, 2023-12-30:
Works on:

```
22.6.0 Darwin Kernel Version 22.6.0: Wed Oct  4 21:25:40 PDT 2023; root:xnu-8796.141.3.701.17~4/RELEASE_ARM64_T8103 arm64
20.6.0 Darwin Kernel Version 20.6.0: Thu Mar  9 20:39:26 PST 2023; root:xnu-7195.141.49.700.6~1/RELEASE_X86_64 x86_64
```
m4 is `GNU M4 1.4.6` from Developer tools in all cases (which it looks like it is in your trace)

(Ventura and Big Sur respectively, I've not had a chance to install dev tools on my Sonoma machine yet)

pauloscustodio, 2023-12-30:
It is failing on the Github workflow, but m4 is used successfully to build the library, so I'm puzzled (zcc.c:1321).

pauloscustodio, 2023-12-31:
Solution was simple: add a newline to the input text.

closed 2023-12-31 by pauloscustodio