z88dk/z88dk #2474: z88dk-win32-2.3.zip is flagged by multiple anti-virus systems

PhilR73, 2024-01-05:
It may well be a false positive, but multiple AV are detecting issues with this zip on Windows. 

See https://www.virustotal.com/gui/file/665fcd1fa9b8f689af6dc3587b51e582759e297f7179ad896624ebc40194b6d5?nocache=1

When unzipping it on my system even Windows Defender (not the most robust AV) flagged issues with 'Wacatac'. File: "z88dk-win32-2.3\z88dk\src\z80asm\build_ldflags2229334.exe"

It would be worth double checking your build system is not compromised, just in case.

suborb, 2024-01-08:
I went through this with @pauloscustodio a couple of months ago (see https://www.virustotal.com/gui/file/bdd2bbf33af084848a52e48a3978d0154536e588e7629b245949b2348ab35ca9?nocache=1 for a report from back then.

We went through binary by binary:

appmake:

https://www.virustotal.com/gui/file/e60126f3211d19fac2a4f5dc371cb0dab6c9a34cd9c993bd043f592c218b53ab

basck:

https://www.virustotal.com/gui/file/f997efcb17dc1b613f370d0d60d4b78169acebbd9802efe7bbea0a0ea6f242ce

copt:

https://www.virustotal.com/gui/file/44b098ce754ec30acf55c857bf168264056c5329b12ff5d74598805d72458f4e

z88dk-dzx0:

https://www.virustotal.com/gui/file/0d8c3c3485ff5354434f3cff4eee104608817b7379c95d48376a362bf98dc225

z88dk-dzx7:

https://www.virustotal.com/gui/file/d12ec2990ee046cdb67c8ae67e40878bddad8f2716930f742aca7cdef7fe5087/detection

z88dk-z80nm:

https://www.virustotal.com/gui/file/1ff934dca381188858293493af42ec792ec425e45d76ac00e79c405d3ab24c7c

z88k-fontpv1000:

https://www.virustotal.com/gui/file/0151b131b1ebf10dc3c015114057e9d620563669e85b231f5025a52c417604f7

z80asm:

https://www.virustotal.com/gui/file/479a3ec32385b5594921d11d33c27bb36c070d74e4a1f5deafc381641e6c7697

We concluded that it was just false positives being picked up by ML scanners.

The binaries are cross-compiled using mingw on Linux

pauloscustodio, 2024-01-09:
```build_ldflags2229334.exe``` looks suspicious, as it is not generated by the tool-chain. The original file is ```build_ldflags.pl```

suborb, 2024-01-09:
I think it must be some debris from the C++ filesystem detection that's not being cleaned:

```
-rwxrwxr-x  1 build build 287222 Jan  9 03:15 build_ldflags1306077.exe
-rwxrwxr-x  1 build build 287222 Jan  9 03:15 build_ldflags1306085.exe
-rwxrwxr-x  1 build build 287222 Jan  9 03:15 build_ldflags1306094.exe
-rwxrwxr-x  1 build build 287222 Jan  9 03:15 build_ldflags1306109.exe
```

suborb, 2024-02-13:
This has reared its head again - between this and Russian spammers on the forum I'm starting to wonder why I bother.

I've reviewed again, and I can see we have the following:

* Most detections are by relatively unknown checkers
* Most detections are by engines that use ML
* Most detections are ML matches
* The files with the most detections are z88dk-zx0 and z88dk-zx7 which are < 250 lines of code
* Recompiling on my desktop also results in detections (admittedly fewer)

My desktop uses a different version of mingw (nightly is a 9.3 from 2020)

So pulling on that thread, it appears we're not alone:

* https://www.reddit.com/r/Malware/comments/erricw/binaries_generated_by_mingw32_detected_as_trojan/
* https://security.stackexchange.com/questions/229576/program-compiled-with-mingw32-is-reported-as-infected
* Issue 10295 on MINGW-packages

amongst many others.

So, options:

* Discontinue Windows builds
* Upgrade mingw on the build machine and hope that it's not a wasted effort
* Start reporting these false positives to the AV vendors
* Use GitHub to build nightly packages for Visual Studio, upload binaries, let users combine binaries + libraries.
* Only release Windows builds with manually built Visual Studio binaries
* Live with it

pauloscustodio, 2024-02-14:
In my opinion, discontinuing Windows builds, or making it more difficult for users is not desirable. I can help on upgrading mingw on the build machine. Ideally we should do it on a clone, to be able to roll-back easily if it does not work as expected. Please let me know.

suborb, 2024-02-15:
I wasn't being entirely serious about stopping building it!

Re-building the container is a pain - mainly gathering up the dependencies - my main concern is that even when it's done, some heuristic will perceive the binaries to be a threat - it seems to be a systemic result of building with mingw which triggers a false positive.

JohnKozell, 2024-03-07:
I'm new to z88dk and received a similar virus notification. Not having any experience yet with z88dk so I was about to make knee-jerk reaction and remove it, thinking it was some cover for a virus scheme. But instead I was led to this post. I'm glad it was here as a known (and more importantly, unwanted) issue. I just wanted to mention this for context with other new-comers.
Can we just omit or change the implementation of this executable?
[image]