z88dk/z88dk #2475: make compatible with casio 8080 CPU

miguelhermida, 2024-01-05:
Casio Z-1,Z-1GR and FX-890p have ASM and C c language support, and have Intel8080 CPU, you can also install FX-DOS from this page:
http://stear.c.ooco.jp/fx890p/fx_fxdos.htm
Could you make z88 compatible so it can be used on these devices?

suborb, 2024-01-10:
It looks like those models use an X86-compatible 16-bit CPU 80L188EB so I think https://github.com/tkchia/gcc-ia16 is probably the right starting point.

zx70, 2024-01-16:
IIRC the old MSDOS Turbo C can be used to cross-develop code on those calculators

https://community.casiocalc.org/topic/643-turbo-c/

miguelhermida, 2024-01-16:
hello good.
I have programmed in C for the Z-1 and Z-1GR in:
 Pacific C 7.51
Turbo C 3.0
Turbo PAscal 7
And in all of them I encounter the same problems, for example I cannot use:
clrscr();
OR
system(“cls”);
OR
system(“clear”);
or ASCII CODE
printf '\033[2J'
I can't run either
gotoxy()
and I can't use any function of the graphics library either.
line,
circle,
etc
What I can do is, for example, a text game but nothing more.
compile compiles without errors but does not execute the functions mentioned above, so I wanted to see if it is possible to make them work on Z88DK.
I think they don't work because it is an LCD screen,
stear has a graphics API that interacts with the calculator but it is in ASM code and I have no idea how to call it in C and, for example, create a sprite or a square.
Here is the link of the api:

Graphics API Manager Ver1.00

http://stear.c.ooco.jp/fx890p/file/GAPI100.LZH

zx70, 2024-02-29:
Working on a z88dk compatible function set sounds like an interesting project, but it is only loosely related.

closed 2024-02-29 by zx70