z88dk/z88dk #2483: How to have random #asm as a beginning of a banked file?

desertkun, 2024-01-27:
I have this code, meant to have some code be generated on bank 1
```c
#pragma bank 1
#pragma codeseg CODE_1
#pragma bssseg BSS_1
#pragma dataseg DATA_1
#pragma constseg RODATA_1

#pragma asm
    jp ___module_action
    jp ___module_loop
#pragma endasm

void ... ()
```

It, however, does not generate the embedded asm code at the beginning of the binary. In fact it doesn't end up in any section, besides main one.

I have to do this trick to make it work:
```c
#pragma bank 1
#pragma codeseg CODE_1
#pragma bssseg BSS_1
#pragma dataseg DATA_1
#pragma constseg RODATA_1

void module_intro_header() __naked
{
#asm
    jp ___module_action
    jp ___module_loop
#endasm
}
```

Here a function pretends to be one, it's there to just allow embedded asm at the begining. But this is ugly.

Is there a more proper to use `#pragma asm`, I don't want to link other asm files, as that means I will have to introduce another section and make my own custom section map, I'd rather stick to default one.

Any ideas?

suborb, 2024-01-30:
You'd have to put the `SECTION` into the #pragma block.

This use case suggests that I need to add a new preamble section to the banked definitions so that it's guaranteed to be first and independent of the file compilation order.

You probably don't care, but for other people reviewing issues, when using sdcc asm blocks have to be within `__naked` function.

desertkun, 2024-02-03:
Thanks, that solves the issues.

closed 2024-02-03 by desertkun