z88dk/z88dk #2487: Windows zcc binary doesn't work on wine

dfandrich, 2024-02-01:
Running zcc from z88dk-win32-2.3.zip with -v on Wine 6.0.3 on Linux x86_64 shows the last command being run is of the form `type "C:\users\myself\Temp\zcc6211.o" >> "hello.o"`  On wine, this produces a file hello.o of 9 characters (instead of zcc6211.o's 432), presumably because `type` is intended for text files and this is a binary file, with byte 10 being NUL.

This command should probably be changed to `copy /y C:\users\myself\Temp\zcc6211.o hello.o` to copy the file verbatim.

suborb, 2024-02-11:
So it seems that over the years, that windows `type` does support binary whilst wine doesn't. 

`copy` was changed to `type` in #46 (124a0ed791e701afc9cc93cc3e4d8ce22aa02022) and there's been no reported windows failures.

> , it's always going to fail. It's assumed there
> https://github.com/z88dk/z88dk/blob/006a60fb09b854b55832d88f36cb3f0687f974a1/src/zcc/zcc.c#L749

Nope, it's a `filter_out` command which is this https://github.com/z88dk/z88dk/blob/006a60fb09b854b55832d88f36cb3f0687f974a1/src/zcc/zcc.c#L763

zx70, 2024-02-25:
The described behaviour makes me think that the wine environment is not providing the shell commands  for windows, it is simply falling to the Linux ones.
I wonder why you should want to have the windows build of z88dk on Linux?   It is obviously possible to manually get an hybrid build but I suspect there's a lot of intermediate configurations usable only in rather restricted environments.

dfandrich, 2024-02-28:
I'll close this issue and raise it on Wine instead, since it sounds like Windows handles this case fine. I was trying to use the Windows binaries under Wine to perform some automated CI tests, and Linux binaries are pretty scarce (I only know of Fedora packages and they're really old).

suborb, 2024-02-28:
> and Linux binaries are pretty scarce (I only know of Fedora packages and they're really old).

I guess it depends on how you're running your build as to whether they can be used, however both the docker and snap builds are nightly.

dfandrich, 2024-03-11:
I've opened [bug #56381](https://bugs.winehq.org/show_bug.cgi?id=56381) on Wine.

closed 2024-03-11 by dfandrich