z88dk/z88dk #2488: CMake: Detecting C compiler ABI info - failed

Yagodnik, 2024-02-01:
Hi there! I was trying to build project with CMake and ZCC compiler, but I ran into some problems.
Here part of cmake logs:
```
-- Building for: Visual Studio 17 2022
zcc - Frontend for the z88dk Cross-C Compiler
See https://github.com/z88dk/z88dk/wiki/CMake for more information
zcc - Frontend for the z88dk Cross-C Compiler
See https://github.com/z88dk/z88dk/wiki/CMake for more information
-- The C compiler identification is MSVC 19.38.33135.0
-- Detecting C compiler ABI info
zcc - Frontend for the z88dk Cross-C Compiler
See https://github.com/z88dk/z88dk/wiki/CMake for more information
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe
zcc - Frontend for the z88dk Cross-C Compiler
See https://github.com/z88dk/z88dk/wiki/CMake for more information
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe - broken
CMake Error at C:/Program Files/CMake/share/cmake-3.24/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe"
  is not able to compile a simple test program.
...
```

CMake fails to detect ABI information. I tried last release and nightly build but they all failed too. 
My only assumption is that I use Windows instead of Linux in docs examples. Can I fix it or I should use WSL or something like this?

My CMake code:
```cmake
project(test2 C)
add_executable(test2bin main.c)
```

lostbits, 2026-08-17:
```
Win11 Pro (x86-64)
cygwin 3.7.0
gcc 17.0.0
cmake 4.2.1
```

The first tutorial in the cmake tutorial (helloworld) failed with "-- Detecting C compiler ABI info - failed" on my laptop but succeeds on my desktop. I don't have enough information to debug this thing myself, for one, I can't find the CMakeError log. Is there a reference available to help me analyze and fix this?

feilipu, 2026-10-27:
closed by https://github.com/z88dk/z88dk/commit/a11376a6f5755f2848fbe80f2f1c883298e920fc

closed 2026-10-27 by feilipu