z88dk/z88dk #2489: Contributing with fixes in header files

vmorilla, 2024-02-05:
I have come across a few issues involving clang which I would like to fix in order to have a proper development environment in VS Code with Intellisense support.

Some example of the issues are in arch/zxn.h where the SDCC __sft syntax is used. So, instead of 
```c 
// tbblue registry system

__sfr __banked __at 0x243b IO_243B;
__sfr __banked __at __IO_NEXTREG_REG IO_NEXTREG_REG;

__sfr __banked __at 0x253b IO_253B;
__sfr __banked __at __IO_NEXTREG_DAT IO_NEXTREG_DAT;
```

I guess I would need
```c
// tbblue registry system
#ifdef __CLANG

extern unsigned char IO_243B;
extern unsigned char IO_NEXTREG_REG;

extern unsigned char IO_253B;
extern unsigned char IO_NEXTREG_DAT;

#else

__sfr __banked __at 0x243b IO_243B;
__sfr __banked __at __IO_NEXTREG_REG IO_NEXTREG_REG;

__sfr __banked __at 0x253b IO_253B;
__sfr __banked __at __IO_NEXTREG_DAT IO_NEXTREG_DAT;

#endif
``` 

However, I'm hesitant to make a pull request proposal since I don't know which header files I should touch. For instance, for arch/zxn.h I see at least the following duplicated entries:
- [./include/_DEVELOPMENT/clang/arch/zxn.h](/z88dk/z88dk/blob/master/include/_DEVELOPMENT/clang/arch/zxn.h)
- [./include/_DEVELOPMENT/proto/arch/zxn.h](/z88dk/z88dk/blob/master/include/_DEVELOPMENT/proto/arch/zxn.h)
- [./include/_DEVELOPMENT/sdcc/arch/zxn.h](/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sdcc/arch/zxn.h)
- [./include/_DEVELOPMENT/sccz80/arch/zxn.h](/z88dk/z88dk/blob/master/include/_DEVELOPMENT/sccz80/arch/zxn.h)
- [./include/arch/zxn.h](/z88dk/z88dk/blob/master/include/arch/zxn.h)

I have tried to find a solution in the z88dk forum, however, it seems I cannot register to raise any new topic.
Is there anyway I could register?

Thanks!

vmorilla, 2024-02-05:
Any possibility to register alternatively to the forums?

suborb, 2024-02-07:
> Any possibility to register alternatively to the forums?

After a few hours of cleanup and fiddling, registration is open again. Sorry about the delay

closed 2024-02-11 by suborb