z88dk/z88dk #2492: Link .map files from other builds instead of clib/startup/crt0?

desertkun, 2024-02-07:
I have an unusual build setup, where I have a "main" application that has most of the code, clib, startup, crt0.
I also have "modules" that I want to build separately, and plan to place and run at known location (`0x1000`).

So imagine something like this:
main.c
```c
int main()
{
    return a + b;
};
```
That main program gets compiled with `-m` and generates `main.map`, that along other things, has
```
l_gint4sp                       = $7474 ; addr,
```
which is clib code nessessary for basic C library to function (like addition).

Now imagine I want to compile a module
module.c
```c
void module()
{
     int c = d + e;
}
```

That module doesn't have main, startup, clib or crt0. But it is getting executed within context on main application, and all clib functions are still there. It is in the `main.map` that `l_gint4sp` is at `$7474`.

So I need a way to link that .map file into the build process of the module, so all those things get resolved.

Besides writing a python script that would pre-process `.map` and convert it into a bunch of `public l_gint4sp; defc l_gint4sp = 0x7474` for `main.asm` which I "link" by compiling it, what are my options?

As you probable can guess, this is banking in disguise. Only the banks are compiled separately, much like dynamic libraries.

suborb, 2024-02-07:
The z80asm -g option might help you here - it creates a defc file of the labels in a binary.

desertkun, 2024-02-07:
Sounds good, thanks.

closed 2024-02-07 by desertkun