z88dk/z88dk #2497: Wav generation issue + potential solution

cstereo, 2024-02-17:
I tried your solution to compile a test C program for the DAI computer. It was great it took me only a few minutes to learn how to install and output the .bin .dai and. wav files. As I have a program to translate .dai in .wav, I confirm that the .dai file is correct and that is works on MAME. I used your "latest" compiled version on windows.

However, there is a problem with the wav files, probably for all supported computers, as low level bits are not stable (see extraction from audacity below)
From your C code, and by analysing the wav header, I understand the following:
- The wav file saves samples as bytes (char) and not as a short
- At line 841 of appmake.c, a small noise of 1 bit is subtracted to the signal
- I assume that this issue is due to the fact that the Low value (0) becomes High when subtracting 1 (0-1=255).
- Solution (not tested) could be to replace “c-(i&1)” by “(c&0xFE)+(i&1)”

Thank you for your help and for your great project. 

[image]

suborb, 2024-02-17:
@zx70 this change makes sense from a code perspective - but what are the effects on wav file generation for other targets?

cstereo, 2024-02-17:
I have almost no knowledge of this project so it is just a guess. From what I have seen this problem shoud arise for all target computers requiring to generate a 44.1KHz wav file (flag mc_options audio=1 and 22 =0) using raw2wav. Digging a little more, I see that is corresponds to around 20 targets.

zx70, 2024-02-17:
I'm aware this WAV encoding mode is sometimes misunderstood by the SW programs 
Nevertheless it is a perfectly valid one.
I chose it because it permits a very high compression rate and it is usually working in most of the configurations, this is (was) especially useful to create 44100 audio sources and burn a CD.

In my past experiments I haven't met stability issues,  but I noticed that some audio editor didn't like this audio format.
It is perfectly fine to fix it if we're inserting a glitch ourselves, though.    Most of the targets include a "volume" option to address possible issues in the audio transfer of conversion, does the DAI target have it?

By the way I'm more than happy to help investigating on it, probably the first thing to check is whether Audacity (being a key reference for the digital audio stuff) would like this workaround.

cstereo, 2024-02-17:
SOME PRELIMINARY REMARKS

Adding some "noise" may help for the compression and audacity is doing it as
well when performing some transformations.

I have notice that MAME is accepting different format and sampling rates. It
does read char encoding (unsigned byte) but does not accept to save in this
format. Saving in char is therefore a good idea for Mame and to play the
sound from Audacity.

 

I am currently working on the DAI cassette input, which is very fragile when
using a PC to play the wav file. This is particularly the case with the V4
hardware version which has a completely different hardware version from the
more frequent V7 version. The electronic simulation shows that there are
very sensible to the slope of the numeric signal : results will therefore be
very different if you use a 22KHz or 44KHz sampling even at the same volume.
It will also depend on your PC sound card (which may not filter properly
high frequencies). This problem is very likely to be similar for other
computers, depending on hardware, because they were designed to get signals
with very narrow bandwidth (cassette = 12KHz max).

There is no volume gain control on the DAI.

 

CURRENT ISSUE

Byte encoding in your wav uses unsigned byte format, i.e. from 0 to 255
according wav norm
(https://ccrma.stanford.edu/courses/422-winter-2014/projects/WaveFormat/)

 

In appmake.c/raw2wav(),

      c=getc(fpin);  // (a)

      fputc(c-(i&1),fpout); // (b)

 

In dai.c/dai_bit()

      h_lvl = 0xff; // High level before noise (c)

      l_lvl = 0x00 // Low level before noise (d)

 

For the DAI, after subtracting Noise at line (c), encoding of 

*	High level is 255 or 254 (254-1 according to line (b))
*	Low level is 0 or 255 (0-1) alternatively (as depicted in previous
post)

This is problematic.

 

This may not be detected in some situation :

*	At the code level, not all samples are read by the computer and
therefore the wrong high level can easily be invisible to the code (andif it
is the case for the first byte, it is very likely to be the case for all
bytes)

(this depends on the computer, but this is the case for the DAI)

*	At hardware level, this alternance of 1 and 0 may result in a level
which will finally be accepted as a low level. (Cassette input bandwidth is
normally much below 44KHz)

 

I may be wrong but at least for the Dai, code should be changed by setting
l_lvl to 1.

 

If the problem is the same for other computers, i.e. low level is set to
zero, then code should be modified at line (c) (see my previous suggestion).

 

Happy to see that you have the time to solve this issue.

zx70, 2024-02-19:
Thank you for clarifying.  For reasons not related to this project I missed the core problem of the values in the negative side of the waveform overflowing!

It definitely must be fixed, I'm surprised it was never spotted

cstereo, 2024-02-19:
Thank you for your reactivity 😉

zx70, 2024-02-19:
I think the problem is solved, that little bit of noise was added to solve an issue on 2 emulators for the zx spectrum, a "too clean" waveform was confusing them, but I don't remember the exact behaviour.

To me is was reasonable to forecast a similar problem hitting other emulators and I chose to make the change global.  It impacted probably the DAI only, because it was creating a wave audio at its maximum possible volume.

I'm changing both appmake.c (disabling the noise insertion when an overflow must be prevented) and dai.c with a new "loud" appmake option.
We have space for improvements, if you would be available to test it, we could pull in a "fast" and a "dumb" mode

cstereo, 2024-02-19:
I am not sure to understand all what you are planning for the DAI but I will
be happy to test it as soon as I get it.

 

FYI, in case, you do not need to have only 0 & 255 values for the DAI, but
volume has to be close to these values because hardware version before V5
(or V6) requires volume to be around 90% of the maximum of cassette player
(to such extent that you had to use only a few models). There is not such
limitation for MAME.

#2497)

 

<https://github.com/z88dk/z88dk/issues/2497#issuecomment-1952370383> , or
unsubscribe
NB73AVCNFSM6AAAAABDNGIFJ2VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTSNJSGM3TA
MZYGM> .
<https://github.com/notifications/beacon/ADMW5V6YRYZ5IQ7QU7AFC6LYUNB73A5CNFS
M6AAAAABDNGIFJ2WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT

zx70, 2024-02-19:
There's an added option (-Cz--loud) to set the volume to 99%.   The default audio level output is now at about 70%, it is hopefully good for most of the hardware configurations to send the audio output to a tape recorder or an amplifier.
I'm going to insert this same option in the ONDRA target as well.

The "dumb" option is simply a shortcut to pass a binary file to appmake (usually including a loader block and a code block together with extra information for the emulator) and create a wave without expecting a z88dk built program.
We can probably forget about it, I see the ".dai" files are rather simple.

By "fast" I refer to shortening a little the audio periods and push the original loader to its limits, sometimes the loading time can be reduced of an interesting amount without hardware or software changes.   It can give good results with some software driven loader.    Sometimes the "long pulse" can't be altered but the short one allows quite a bit of tweaking, in example.

cstereo, 2024-02-19:
Understood for the volume level option.

 

Apart from compressing the size of the program, loading faster on MAME has
little interest because you can accelerate the DAI CPU and load the full
memory in a few second. 

 

This is more interesting for the physical DAI which effectively can load
much faster some files up to +%60.

Check this link if you want to have more information on this (Cassette
section . . I do not know to which extent you know the DAI.

https://filedn.eu/l8fxcG16N8iQwXfMASdK9c4/DAI/plan.html

This site gives all information you can need on the DAI. Some programs are
provided with the normal and the fast version of the audio file. Bruno, the
owner and main contributor (by large) to the site has programmed a .dai to
.wav converter with an option to accelerate the loading.
https://filedn.eu/l8fxcG16N8iQwXfMASdK9c4/DAI/programmes/wav2dai%20et%20dai2
wav/index.html

 

For audio input, the key rule is that the trailer should not be accelerated,
but you can do it on main data.

However, this can be tricky because in practice it depends on the version of
the DAI (and the dispersion of components.) and to a lesser extent on the
player (ex: PC card). I have a DAI V4 which is not favorable for this. I am
building an upgrade to have the same input has the DAI V7 but it will take a
few weeks. However, I can ask some fellows (who have contributed to the site
above) if they can test your files if you are interested.

 

Impatient to test the results. When are you expecting to push the correction
(on the windows installer preferably) ?

zx70, 2024-02-19:
I hope the nightly build will help us.   If everything goes well we should have a fresh build tomorrow morning (nightly.z88dk.org), it will deliver a portable build for Windows too.

I've already introduced a '--fast' option in the DAI target.  You're right, I know very little of it, but I istinctively avoided changes on the leading and trailing pulses.  I've been cautious, so the little speed-up would save, if it works, only 10 seconds every 1.5 minutes of a standard loader.
Building appmake is not difficult at all, you could experiment with it directly, I think.

I'm quite ashamed for the nast bug, luckily in most of the affected targets it was showing up only with the --loud option.  Others ones weren't affected at all.

zx70, 2024-02-20:
The compression.  A normal non lossy dictionary based or even RLE compression will give better results than mp3 .  Gzip or pkzip should reduce a 5mb wave file to 100k.

cstereo, 2024-02-21:
I built the nightly version using visual studio, which is much simpler than
CMAKE, to release x64 version. I then copied Clibs directory and
copy/changed config.h in src directory. I was able to generate, load on MAME
and execute a program with the different options combinations (-Cz--loud
-Cz--fast)

I think this wav issue can therefore be closed.

 

I tried two small programs that crashed, probably because of overflow issues
(ex: on exit). I think will need to learn a little more about your project
directives to avoid that.

 

BZW: When building zcc for Win32 (but not when building for x64) Bitdefender
detected a trojan, after linking,: \win32\Release\z88dk-sccz80.exe :
Gen:Variant.Fragtor.507608. I already had a discussion with another
contributors (Dom) on this and he told me that a ticket was issued
https://github.com/z88dk/z88dk/issues/2474. When you download your program
to VirusTotal and go to the behavior tab, it suggests that there may still
be a problem.

 

Thank you again!

#2497)

 

RG37AVCNFSM6AAAAABDNGIFJ2VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTSNJTGYZTA
OBXGM> .
<https://github.com/notifications/beacon/ADMW5VZD3AKTMG4ZI26UTKTYURG37A5CNFS

closed 2024-02-25 by zx70