z88dk/z88dk #2500: `libsrc` build system prone to bricking when interrupted

atirut-w, 2024-02-26:
This can be fixed by running `git clean -xdf`, but could be confusing to someone new to this project.

suborb, 2024-03-03:
What does bricking mean in this context?

atirut-w, 2024-03-04:
Failing to build in subsequent runs after being interrupted.

zx70, 2025-06-09:
which system environment are you using?
I got similar problems on Windows with msys, but it's not happening on z88dk only (tbh it's less problematic than other builds)

atirut-w, 2025-06-10:
It happened on Fedora, but I'm not sure if the issue still happens because I haven't tried building it in a while.

zx70, 2025-06-11:
No problem, I know your notice aged in the meanwhile, I'm sure we all kept an eye on it while building the project locally.
If you don't mind I'm closing this one, if it happens again please help us with some more detail (e.g. understanding what was being built when interrupted, the OS environment, etc..).

closed 2025-06-11 by zx70