z88dk/z88dk #2507: (z80asm) Parse1.c compilation issue for m68k

suborb, 2024-03-10:
Initially I thought it was the switch table (I've seen issues with that on m68k in the past), but it might be the length of the function causing a lot of debug information?

```
ccache m68k-atari-mint-gcc -std=gnu11 -MMD -Wall -O3 -ggdb -I. -Isrc -I../common -Isrc/c -It -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -O2 -fomit-frame-pointer -m68020-60 -mlong-jump-table-offsets -O0 -c -o src/c/parse1.o src/c/parse1.c
/tmp/ccUKbNh3.s: Assembler messages:
/tmp/ccUKbNh3.s:487310: Warning: .stabs: description field '10908' too big, try a different debug format
```

Occurred here:

[image]

pauloscustodio, 2024-03-28:
I've been trying to setup m68k-atari-mint-gcc -std=gnu11 for testing with no success; its build from github fails on latest Ubuntu, the binary versions I found need cygwin32 to run and I could not setup cygwin32 on a Windows 10 box.

I need some pointers on how to run m68k-atari-mint-gcc

suborb, 2024-03-28:
nightly uses the binaries downloaded from here: https://tho-otto.de/crossmint.php using the m68k-atari-mint toolchain.

commit 2f0a0781 (2024-03-29): Fix #2507: remove debug info for m68k-atari-mint-gcc
closed 2024-03-29 by pauloscustodio