z88dk/z88dk #2508: Far memory support for MSX

bradstallion, 2024-03-19:
Hi,
is there any plan to implement far memory support for MSX?
Thanks!

suborb, 2024-03-20:
I've not had any thoughts about adding `far` support to any other machines - it was needed on the z88 because the OS was built around 3 byte pointers and provided allocation features (albeit only up to ~250 bytes).

For machines without such restrictions, and access to banks of contiguous memory, I think it's easier to just use [Named address space](https://github.com/z88dk/z88dk/wiki/More-Than-64k#named-address-spaces-classicnewlib-sccz80-and-zsdcc)

bradstallion, 2024-03-21:
Thank you for your answer. I've read your link. I'm developing for MSXDOS2, so I know how to manage the memory mapper by calling EXTBIO. But it's not clear to me the meaning of:

```
void setb0(void);            // The function that sets the currently active memory bank to b0
void setb1(void);            // The function that sets the currently active memory bank to b1
```

Moreover, I tried the [linked](https://github.com/z88dk/z88dk/tree/master/examples/named-address-spaces) example, but it seems that both files (bank1 and bank4) end on the same bank:

[image]

The first line ("Value in address space 1 is..") should be 89.
Thanks

bradstallion, 2024-03-21:
... my fault, I was using the emulator in 48k mode. With 128k it works!
Anyway, can you explain "sets the currently active memory bank to b0" for MSX? Thanks

suborb, 2024-03-21:
It means that we simply page in that bank into the memory space. In that example, I named the address spaces after the bank indices because I thought it was simpler and tied in with the general way banking is done within z88dk

For the MSX2, you'll need to maintain a map of physical to logical banks. So for allocation you've probably got code like this:

```
    ld      b,0             ;Primary mapper
    ld      a,0             ;Allocate user segment
    call    ALL_SEG
    jp      c,print_message_pop
    ; a = allocated segment
```

And then you'll need to keep track of logical to physical map:

```
    ; e = logical bank
    ; c = physical bank
    ld      hl,bank_mappings
    ld      d,0
    add     hl,de
    ld      (hl),c         ;Save mapping

bank_mappings: defs 256
```

So to switch to implement setb0 and setb1 the code would look like this:

```
_setb0:
    ld    a,(bank_mappings)
    call    PUT_P2  ;page the bank into segment2
    ret

_setb1:
    ld    a,(bank_mappings+1)
    call    PUT_P2  ;page the bank into segment2
    ret
```

`ALL_SEG` and `PUT_P2` are obtained from EXTBIO:

```
    xor     a
    ld      de,$0401
    call    $FFCA   ;EXTBIO
    xor     a
    ld      de,$0402
    call    $FFCA   ;EXTBIO
    and     a
    ld      de,msg_nomapper
    jp      z,print_message
    ld      de,dos2_jump
    ld      bc,dos2_end - dos2_jump
    ldir
```

The code snippets have been pulled out from lib/target/msx/classic/msxdos.asm - if you use the pre-existing banking support it will load code/data into the appropriate banks and maintain the logical-physical mapping for you. So all you'll need to do is to provide setb0, setb1 and add a cheeky `PUBLIC bank_mappings` to msxdos.asm (I'll export an identifier by default so you won't need to do that forever)

bradstallion, 2024-03-21:
Thanks a lot!

suborb, 2024-04-09:
To be honest, as you can probably guess I completely forgot!

bradstallion, 2024-04-09:
No problem! I hope you'll add it, when you'll have the time! :-D

suborb, 2024-04-09:
Just reopening so I don't forget again!

suborb, 2024-04-12:
Sorted - hopefully your version looks at least somewhat similar!

bradstallion, 2024-04-13:
That's great! Thanks a lot!

closed 2024-04-13 by bradstallion