z88dk/z88dk #2509: [appmake][c128] Capital/Small generated file names

Fabrizio-Caruso, 2024-03-19:
I need to build my games under any POSIX environment including both Cygwin/Windows and Linux.

For the c128 I would like to generated disk images (which I can sort of do with the c1541 tools distributed with Vice emulator).

The problem I have is that appmake is not generating the same file names when run under Cygwin/Windows vs Linux.

Under windows appmake with `-create-app -o"a40.bin"` generates `a40.LDR` and `a40` (and `a.bin`) whereas under Linux I get  `A40.LDR` and `A40` (and `a.bin`). I could use a conditional to check that I am not under Cygwin/Windows and rename them to `a40.LDR` and `a40` but it looks like an ugly solution and the behavior of appmake seems wrong to me.

My makefile target is:
```
c128_z80_40col: $(ASSETS_PATH)/z88dk_xchase.asm 
	$(Z88DK_PATH)$(MYZ88DK) +c128 $(SCCZ80_OPTS)  \
	$(INCLUDE_OPTS) \
	-lndos -subtype=disk -clib=gencon \
	-D__C128_Z80__ -DXSize=40 \
	 \
	-D__BIT_BANG_SOUND   \
	-D_XL_SLOW_DOWN_FACTOR=$(_COMMON_SLOWDOWN_FACTOR)*$(_C128_Z80_40COL_SLOWDOWN) \
	-D__CONIO_GRAPHICS  \
	$(FULL_FILES) \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	-create-app -o"a40.bin"
ifdef USE_TOOLS
	$(C1541) -format "crosschase,0" d64 FULL_$@.d64
	$(C1541) -attach FULL_$@.d64 -write a40.LDR
	$(C1541) -attach FULL_$@.d64 -write a40
	mv FULL_$@.d64 $(BUILD_PATH)/X$(GAME_NAME)_$@.d64
	rm a40.bin
else
	mv a40.bin $(BUILD_PATH)/X$(GAME_NAME)_$@.bin
endif
	rm a40.LDR
	rm a40
```

suborb, 2024-03-19:
Actually, the filename is intended to be uppercase: https://github.com/z88dk/z88dk/blob/master/src/appmake/c128.c#L71

So, it's Cygwin causing a bit grief - can you do a compile with -v so I can see what commands are being run?

Fabrizio-Caruso, 2024-03-19:
@suborb but what appmake generates under Linux won't work with c1541 unless renamed to small letters because the loader expects a non-shifted file name in the disk image (non-shifted is shown either as small letters in small+capital char set or as capital in capital+graphics depending on the Petscii mode). 

Do you really mean that `-create-app -o"a40.bin"` is supposed to generate capitalized `A40` and `A40.LDR` where the `A40.LDR` expects a small letter file name?

Fabrizio-Caruso, 2024-03-19:
I think the problem is that appmake assumes small (un-shifted PETSCII) letters in the file name to be used in the loader:
```
z88dk-appmake +c128 --disk  -b "a40.bin" -c "a40"
c1541 -format "crosschase,0" d64 FULL_c128_z80_40col.d64
formatting in unit 8 ...
c1541 -attach FULL_c128_z80_40col.d64 -write "a40.LDR"
cannot read file `a40.LDR': No such file or directory
cannot read file
make: *** [Makefile_common:858: c128_z80_40col] Error 1
New binaries built: 0
```
but it generates capitalized files, which once put inside the disk images won't match with the loader.
Indeed I can load the game if I edit the BASIC loader.

For some unknown reason, I don't have this problem in Cygwin/Windows where appmake seems to follow the capital/small type passed with `-o`.

suborb, 2024-03-19:
> For some unknown reason, I don't have this problem in Cygwin/Windows where appmake seems to follow the capital/small type passed with `-o`.

I think we're talking slightly at cross purposes. It looks like appmake is deliberately uppercasing the filename. This uppercased version is then used in the loader and the filename.

It's not a solution, but the following is a hint as to what's going on.

I've managed to recreate something similar on macOS which has a case-insensitive but preserving filesystem as does windows. Compare the outputs of the following commands:

```
zcc +c128 world.c -create-app -o boobar.bin  -Cz-d
```

Yields: boobar.bin BOOBAR and BOOBAR.LDR

```
zcc +c128 world.c -create-app -o bobber -Cz-d
```

Yields: boobar and BOOBAR.LDR

In both cases, BOOBAR.LDR contains "BOOBAR" as the filename - that should be the same for you.

There's obviously something going wrong here - I think we might have to force an `remove` before writing the output file to sort out these ambiguities.

Fabrizio-Caruso, 2024-03-19:
If I do as above, under Cygwin/Windows:
```
c128_z80_40col: $(ASSETS_PATH)/z88dk_xchase.asm 
	$(Z88DK_PATH)$(MYZ88DK) +c128 $(SCCZ80_OPTS)  \
	$(INCLUDE_OPTS) \
	-lndos -subtype=disk -clib=gencon \
	-D__C128_Z80__ -DXSize=40 \
	 \
	-D__BIT_BANG_SOUND   \
	-D_XL_SLOW_DOWN_FACTOR=$(_COMMON_SLOWDOWN_FACTOR)*$(_C128_Z80_40COL_SLOWDOWN) \
	-D__CONIO_GRAPHICS  \
	$(FULL_FILES) \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	-create-app -o a40 -Cz-d
ifdef USE_TOOLS
	$(C1541) -format "crosschase,0" d64 FULL_$@.d64
	$(C1541) -attach FULL_$@.d64 -write A40.LDR
	$(C1541) -attach FULL_$@.d64 -write A40
	mv FULL_$@.d64 $(BUILD_PATH)/X$(GAME_NAME)_$@.d64
else
	mv A40 $(BUILD_PATH)/X$(GAME_NAME)_$@.bin
endif
	rm A40.LDR
	rm A40
```
I get an error with the loader:
[image]

Fabrizio-Caruso, 2024-03-19:
P.S.: In standard PETSCII mode a capital file name should look like this (I have edited manually):
[image]

Fabrizio-Caruso, 2024-03-20:
@suborb, I get the same problem under Unix.

appmake is storing unshifted file names in the loader. 
Remark: "unshifted" looks like "capital" letters in the standard PETSCII mode, which maybe why you say it is capital letters, but it is not. It is basically a small letter file name, which is not correct.

suborb, 2024-03-20:
Right, I've finally setup Vice and I can see what's going on and it's a little painful.

* So the filename in the loader  is correct - it's using code points 0x41 -> 0x5a to represent the letters.
* The output filenames should be in lower-case or c1541 prints garbage when you attach them
* The actual load line should be `BLOAD "[filename]"` not `BLOAD "[filename]",B0` - it's that ,B0 that triggers the error it seems.

Fabrizio-Caruso, 2024-03-20:
Yes, Cygwin for some reason was forcing the binary files to be in lower case and so it worked with the loader by coincidence.

Any way, I can now generate upper case binaries on both Cygwin and Linux. So if your latest commits aligns the binary with the loader, everything should work fine.

Fabrizio-Caruso, 2024-03-21:
Tested KO under Cygwin. It is now a bit worse because -o a40 produces an error about input and output file having the same name. So I must use -o a40.bin.

My Makefile target is:
```
c128_z80_40col: $(ASSETS_PATH)/z88dk_xchase.asm 
	$(Z88DK_PATH)$(MYZ88DK) +c128 $(SCCZ80_OPTS)  \
	$(INCLUDE_OPTS) \
	-lndos -subtype=disk -clib=gencon \
	-D__C128_Z80__ -DXSize=40 \
	 \
	-D__BIT_BANG_SOUND   \
	-D_XL_SLOW_DOWN_FACTOR=$(_COMMON_SLOWDOWN_FACTOR)*$(_C128_Z80_40COL_SLOWDOWN) \
	-D__CONIO_GRAPHICS  \
	$(FULL_FILES) \
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	-create-app -o a40.bin -Cz-d
ifdef USE_TOOLS
	$(C1541) -format "crosschase,0" d64 FULL_$@.d64
	$(C1541) -attach FULL_$@.d64 -write A40.LDR
	$(C1541) -attach FULL_$@.d64 -write A40
	mv FULL_$@.d64 $(BUILD_PATH)/X$(GAME_NAME)_$@.d64
	rm a40.bin
else
	mv a40.bin $(BUILD_PATH)/X$(GAME_NAME)_$@.bin
endif
	rm A40.LDR
	rm A40
```

The compilation works but the file name in the loader is the same as before (i.e., unshifted = lower case in alternate PETSCII mode even though it is shown as upper case in standard PETSCII mode).
[image]

I can make the loader work if I edit "A" -> shifted A (spade character in standard PETSCII mode)

[image]

suborb, 2024-03-21:
So, I've no idea what's going on here - your project works for me after a small change:

```
@@ -840,7 +840,7 @@ c128_z80_40col_zsdcc: $(ASSETS_PATH)/z88dk_xchase.asm
        -create-app -o"a40.bin"
 ifdef USE_TOOLS
        $(C1541) -format "crosschase,0" d64 FULL_$@.d64
-       $(C1541) -attach FULL_$@.d64 -write a40.LDR
+       $(C1541) -attach FULL_$@.d64 -write a40.ldr
        $(C1541) -attach FULL_$@.d64 -write a40
        mv FULL_$@.d64 $(BUILD_PATH)/X$(GAME_NAME)_$@.d64
        rm a40.bin
@@ -867,7 +867,7 @@ c128_z80_40col: $(ASSETS_PATH)/z88dk_xchase.asm
        -create-app -o"a40.bin"
 ifdef USE_TOOLS
        $(C1541) -format "crosschase,0" d64 FULL_$@.d64
-       $(C1541) -attach FULL_$@.d64 -write a40.LDR
+       $(C1541) -attach FULL_$@.d64 -write a40.ldr
        $(C1541) -attach FULL_$@.d64 -write a40
        mv FULL_$@.d64 $(BUILD_PATH)/X$(GAME_NAME)_$@.d64
        rm a40.bin
```

```
Volumes/vice-arm64-gtk3-3.8/vice-arm64-gtk3-3.8/bin/c1541 -format "crosschase,0" d64 FULL_c128_z80_40col.d64
/Users/dom/z88dk/CROSS-LIB/src
formatting in unit 8 ...
/Volumes/vice-arm64-gtk3-3.8/vice-arm64-gtk3-3.8/bin/c1541 -attach FULL_c128_z80_40col.d64 -write a40.ldr
/Users/dom/z88dk/CROSS-LIB/src
writing file `A40.LDR' as `A40.LDR' to unit 8
/Volumes/vice-arm64-gtk3-3.8/vice-arm64-gtk3-3.8/bin/c1541 -attach FULL_c128_z80_40col.d64 -write a40
/Users/dom/z88dk/CROSS-LIB/src
writing file `A40' as `A40' to unit 8
mv FULL_c128_z80_40col.d64 ../build/Xchase_c128_z80_40col.d64
```

[image]

[image]

[image]

```
% hexdump -C ../build/Xchase_c128_z80_40col.d64
...
00015000  00 2a 01 1c 09 1c 0a 00  fe 02 30 00 17 1c 14 00  |.*........0.....|
00015010  fe 11 22 41 34 30 22 2c  42 30 00 17 1c 32 00 9e  |.."A40",B0...2..|
00015020  32 30 34 38 30 3a 80 00  00 00 00 00 00 00 00 00  |20480:..........|
00015030  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
...

00016530  15 ff ff 1f 15 ff ff 1f  15 ff ff 1f 01 00 00 08  |................|
00016540  00 00 00 00 00 00 00 00  11 fc ff 07 13 ff ff 07  |................|
00016550  13 ff ff 07 13 ff ff 07  13 ff ff 07 13 ff ff 07  |................|
00016560  13 ff ff 07 12 ff ff 03  12 ff ff 03 12 ff ff 03  |................|
00016570  12 ff ff 03 12 ff ff 03  12 ff ff 03 11 ff ff 01  |................|
00016580  11 ff ff 01 11 ff ff 01  11 ff ff 01 11 ff ff 01  |................|
00016590  43 52 4f 53 53 43 48 41  53 45 a0 a0 a0 a0 a0 a0  |CROSSCHASE......|
000165a0  a0 a0 30 20 a0 32 41 a0  a0 a0 a0 00 00 00 00 00  |..0 .2A.........|
000165b0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00016600  00 ff 82 11 00 41 34 30  2e 4c 44 52 a0 a0 a0 a0  |.....A40.LDR....|
00016610  a0 a0 a0 a0 a0 00 00 00  00 00 00 00 00 00 01 00  |................|
00016620  00 00 82 11 01 41 34 30  a0 a0 a0 a0 a0 a0 a0 a0  |.....A40........|
00016630  a0 a0 a0 a0 a0 00 00 00  00 00 00 00 00 00 3d 00  |..............=.|
00016640  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
0002ab00
```

Can you attach the t64 and the two a40 files you're generating, and I'll compare them.

Fabrizio-Caruso, 2024-03-22:
Thanks! I have now been able to test your solution after a minor change to my Makefile.
It now works perfectly on both Cygwin/Windows and Ubuntu (Lubuntu 21.10) with the very same Makefile.

closed 2024-03-22 by Fabrizio-Caruso