z88dk/z88dk #2510: [snap][bug?] How to install the latest z88dk under Ubuntu 21.10?

Fabrizio-Caruso, 2024-03-22:
I have Lubuntu 21.10 and I would like to install the very latest version of Z88DK.
It is no longer impossible to do it from source because some dependencies are missing or impossible to retrieve.

Therefore I have tried snap, which is supposed to use the very latest version from source, but zcc still
```
sudo snap install --edge z88dk
```
which terminates happily declaring to have install the very latest version from 2024 but when I run `zcc` I get the previously installed version.

Where does snap install z88dk? How can I run z88dk when installed from snap?

suborb, 2024-03-22:
Regarding the regular install, have you switched your repos over to `old-releases.ubuntu.com`? Re-run apt-get update and you should be able to install (I'm guessing) libgmp3-dev - I've just tried it on an 18.10 machine.

The snap install details are this page: https://github.com/z88dk/z88dk/wiki/Snap-usage

Basically, everything ends up being namespaced as `z88dk.[command]` which means you end with the bulky `z88dk.z88dk-z80asm`  so you run the `snap alias` command to make them match a regular install.

Fabrizio-Caruso, 2024-03-22:
@suborb Thanks for your reply. There is no `z88dk-zcc`, which is what I need... :-( 
I see the other binaries with the `z88dk-` prefix but not `zcc`.

suborb, 2024-03-22:
Nope, there's no `z88dk-zcc`, we've never changed the name so it's always been `zcc`. In the snap package there is `z88dk.zcc`

```
dom@florence:/snap/z88dk/current/bin$ pwd
/snap/z88dk/current/bin
dom@florence:/snap/z88dk/current/bin$ ls
z80asm          z88dk-basck  z88dk-dzx7         z88dk-sccz80  z88dk-z80nm     z88dk-zsdcc
z88dk-appmake   z88dk-copt   z88dk-font2pv1000  z88dk-ticks   z88dk-z80svg    z88dk-zx0
z88dk-asmpp     z88dk-dis    z88dk-gdb          z88dk-ucpp    z88dk-zobjcopy  z88dk-zx7
z88dk-asmstyle  z88dk-dzx0   z88dk-lib          z88dk-z80asm  z88dk-zpragma   zcc
dom@florence:/snap/z88dk/current/bin$ z88dk.z
z88dk.z88dk-appmake      z88dk.z88dk-dzx7         z88dk.z88dk-z80svg
z88dk.z88dk-asmpp        z88dk.z88dk-font2pv1000  z88dk.z88dk-zobjcopy
z88dk.z88dk-asmstyle     z88dk.z88dk-lib          z88dk.z88dk-zx0
z88dk.z88dk-basck        z88dk.z88dk-ticks        z88dk.z88dk-zx7
z88dk.z88dk-dis          z88dk.z88dk-z80asm       z88dk.zcc
z88dk.z88dk-dzx0         z88dk.z88dk-z80nm
```

Which, if you follow the wiki page can then be aliases to `zcc` so to match a regular installation:

```
dom@florence:/snap/z88dk/current/bin$ snap aliases
Command                      Alias               Notes
z88dk.zcc                    zcc                 manual
dom@florence:/snap/z88dk/current/bin$ which zcc
/snap/bin/zcc
```

Fabrizio-Caruso, 2024-03-22:
Thanks for your reply. I have not switched to the `old-releases.ubuntu.com`.

I found `z88dk.zcc`. Sorry for missing it.

Also thanks for pointing me to `snap alias`

suborb, 2024-03-22:
Glad you're all sorted!

closed 2024-03-22 by suborb