z88dk/z88dk #2511: Compiler and linker hints mess with Eclipse and probably other IDEs

sgjava, 2024-03-23:
So, as I'm building out my lib and adding assembler routines I'll have something like this in my header:

`extern void __LIB__ playPcm4Sid(unsigned char *buffer, unsigned int len) __smallc;`

This totally hoses the Eclipse Organize imports, format, ctrl+click, etc. To work around this I use sed magic in my build script to remove the comment portion from headers before invoking build:

`extern void /*__LIB__*/ playPcm4Sid(unsigned char *buffer, unsigned int len) /*__smallc*/;`

It would be nice to put z88dk specific compiler/linker directives in comments. Unless others are using an IDE that understands this stuff.

suborb, 2024-03-23:
If you include `<sys/compiler.h>` at the top then it can be waved away - there's handling in there for Clion + Intellisense editors - I'm sure we can add another case!

sgjava, 2024-03-23:
So <sys/compiler.h> removes the need for those directives? I assume I only need to add that to function source I want to add to a lib? i.e. not programs.

suborb, 2024-03-23:
No, they're needed for programs too, otherwise the calling and linking won't work.  However, when working in an editor they're not needed.

We've run into your issue in the past, so at the top of that file we have:

```
#if defined(__CLION_IDE__) | defined(__INTELLISENSE__)

#define __LIB__
#define __SAVEFRAME__
#define __z88dk_fastcall
#define __FASTCALL__
#define __CALLEE__
#define __SCCZ80
#define __Z80
#define __naked
#define __z88dk_callee
#define __stdc
#define __smallc
#define __preserves_regs
#define __no_z88dk_declspec
#define __at(x)
#define __sfr
#define __vasmallc
```

just add `| defined(XXX)` where `XXX` is whatever #define Eclipse sets.

sgjava, 2024-03-24:
So in this case https://github.com/z88dk/z88dk/blob/f96839b5f357e01dc315ac92f3625542cd663954/include/c128/sid.h#L13 the __LIB__ is not required (if using one of the IDEs you pointed out)?

sgjava, 2024-03-24:
OK, so what I did for now was right-click on your project in the “Project Explorer” pane on the left –> Properties –> C/C++ General –> Paths and Symbols –> Symbols tab –> select either GNU C or GNU C++ –> click the Add button at the top-right –> type **__CLION_IDE__** for name and **1** for value.

If you feel like it maybe add __ECLIPSE_IDE__ since Eclipse doesn't define anything unique I can see.

[image]

sgjava, 2024-03-24:
Closing this out, thanks again.

closed 2024-03-24 by sgjava