z88dk/z88dk #2512: [feature request][devcontainer] No install option

Fabrizio-Caruso, 2024-03-25:
It could be possible to use z88dk (some sort of sandbox) on a browser running a dockerized installation for free by just adding a devcontainer folder. No need to explicitly install anything. People would just have to clone z88dk and use codespaces on a browser (for free with a monthly time limit) or locally in vscode. For codespaces there is not even the need for a docker engine. A browser is enough.

Something like the following folder would suffice:
https://github.com/Fabrizio-Caruso/CROSS-LIB/tree/master/.devcontainer

Could such a feature be added?
I am not an expert but maybe what I use for Cross-Lib (which includes z88dk and other compilers) would just work straightway as I already do this for a superset of z88dk.

suborb, 2024-04-27:
I'm not sure how useful it actually is (since it's setup to edit z88dk code!) but as an example of how to leverage the z88dk docker image it's probably quite handy.

Fabrizio-Caruso, 2024-04-27:
You are right indeed: the Z88DK repo is meant to be used to develop Z88DK itself.
It does make more sense for my own project which is just used as is (as source files) and it depends on several tools and compilers (including Z88DK) but even just Z88DK could let people give it.

closed 2024-04-27 by Fabrizio-Caruso

suborb, 2024-04-27:
I've created an example project which makes things a lot clearer: https://github.com/z88dk/z88dk-devcontainer-example