z88dk/z88dk #2513: Docker images should be tagged

dfandrich, 2024-03-26:
The only tag currently available for [z88dk/z88dk](https://hub.docker.com/r/z88dk/z88dk/tags) images on Dockerhub is "latest". It would be very useful to have tags based on the source code release versions (e.g. "2.3") so different compiler versions can be easily tested and builds can be made repeatable.

jorgegv, 2024-04-26:
@suborb +1 to this, I also would need it to setup my CI github actions for RAGE1

jorgegv, 2024-04-26:
It would also be great if you pushed images for the existing tags, there are not so many... 4-5.... ;-)

TIA

suborb, 2024-04-26:
> It would also be great if you pushed images for the existing tags, there are not so many... 4-5.... ;-)
> 
> TIA

I  was hoping re-pushing the tags and your PR would do the trick.

Sadly not so I'll have to do a manual build and push.

jorgegv, 2024-04-26:
ah, yes it seems it does not work for alreadu pushed tags... Pity

jorgegv, 2024-04-26:
Anyway, thanks for the blazing fast response :-)

suborb, 2024-04-26:
nw, you caught me at the right time!

suborb, 2024-04-27:
I've pushed 2.1, 2.2, 2.3 builds.

Unfortunately 2.0 and 1.99c aren't buildable due to symbol resolution issues:

```
8.149 /usr/lib/gcc/x86_64-alpine-linux-musl/13.2.1/../../../../x86_64-alpine-linux-musl/bin/ld: m5.o:/opt/z88dk/src/appmake/m5.c:18: multiple definition of `checksum'; sc3000.o:/opt/z88dk/src/appmake/sc3000.c:24: first defined here
```

dfandrich, 2024-04-27:
Thank-you!

Adding -Wl,--allow-multiple-definition to LDFLAGS should get over the multiple
definition error, but I personally am not interested in using the older
versions at the moment.

jorgegv, 2024-04-27:
Thanks @suborb !

suborb, 2024-04-27:
> Thank-you! Adding -Wl,--allow-multiple-definition to LDFLAGS should get over the multiple definition error, but I personally am not interested in using the older versions at the moment.

Nice to see that it's not just z88dk where there's always an option!

Sorted, pushed those two as well for the archaeologists.

closed 2024-04-27 by suborb