z88dk/z88dk #2521: (z80asm) Pseudo instructions causing trouble

suborb, 2024-04-20:
Specifically a conversion for the gbz80 io instructions:

```
        ld      (0xff00),a
```

```
z88dk-z80asm -v x
Predefined constant: __CPU_Z80__ = 1
Predefined constant: __CPU_ZILOG__ = 1
Predefined constant: __FLOAT_GENMATH__ = 1
Library 'z88dk-z80asm.lib' not found
Reading library '/Users/dom/z88dk/lib/z88dk-z80asm.lib'
Assembling 'x.asm'
x.asm:3: error: illegal identifier
  ^---- ld (0xff00),a
      ^---- ldh(0),a
```

suborb, 2024-04-21:
Sorry @pauloscustodio it's a more general problem:

```
        ld      (0xff00),a
        ld      a,(0xff00)
        ld      (0xff00 + 2),a
        ld      a,(0xff00 + 2)
```

which are all valid z80 opcodes.

yields:

```
z80asm -b x
x.asm:3: error: illegal identifier
  ^---- ld (0xff00),a
      ^---- ldh(0),a
x.asm:4: error: illegal identifier
  ^---- ld a,(0xff00)
      ^---- ldh a,(0)
x.asm:5: error: illegal identifier
  ^---- ld (0xff00 + 2),a
      ^---- ldh(2),a
x.asm:6: error: illegal identifier
  ^---- ld a,(0xff00 + 2)
      ^---- ldh a,(2)
```

closed 2024-04-22 by pauloscustodio

pauloscustodio, 2024-04-22:
Done as suggested, thanks for reporting.

suborb, 2024-04-22:
Thanks @pauloscustodio 

Hopefully there aren't any more like that.

commit f3575acd (2024-04-21): Fix #2521: do not translate ldh instructions if not on a gbz80 cpu
commit 99452574 (2024-04-21): Fix #2521: do not convert "ld (0xff00)" to "ldh", only "ld (0xff00+N)"
commit 75aad260 (2024-04-21): Fix #2521: do not convert "ld (0xff00)" to "ldh", only "ld (0xff00+N)"
commit b09b45fa (2024-04-22): Fix #2521: only convert "ld (0xff00+N)" to "ldh" on gbz80