z88dk/z88dk #2525: 2.2 does not compile on Fedora 40

jorgegv, 2024-04-29:
I tried to recompile v2.2 under Fedora 40 (just released), and it does not build. It builds fine in Fedora 39.

The error is due to a missing initializaton in `src/appmake/zxn.c`, line 34 (field `usr_address`), and it was fixed in commit 3726216ff26, but this was after 2.2 was released. It seems the upgraded GCC in F40 catches the error and the previous version in F39 does not.

Since v2.2 has already been released and the error is fixed in later versions, I think it would be needed to put out a new v2.2.1 with the backported fix just for that line (it's the only error in the build), so that 2.2.x can still be compiled with modern, stricter compilers.

suborb, 2024-04-29:
Is this just a cunning plan to test out the tagged docker pipeline?

In all seriousness, I'm not particularly keen on doing a release of 2.2.1:

* There are some real clangers in 2.3 that are arguably higher priority
* Building all the artefacts manually will be time consuming for me (the nightly machine is just setup to track master)

However, if it fits in with your workflow, I can create a release-2.2 branch, that commit can be cherry-picked in (and probably fixed up) and you can pull that branch and build.

jorgegv, 2024-04-29:
Oh, never mind for me, I have already documented it for myself and I already fixed Cesare to compile with 2.3 (with 2.3 a binary took just 2 bytes more than 2.2, and it broke my build!)

I thought making a 2.2.1 release and image would be simple, since it's only a one time job, fire and forget.

Point is, 2.2 will cease to compile in any up to date machine in the near future... Fedora has been among the first to update the compiler, but the rest will follow shortly.

suborb, 2024-04-29:
> I thought making a 2.2.1 release and image would be simple, since it's only a one time job, fire and forget.

For a regular release it is! I just promote the last nightly.

> Point is, 2.2 will cease to compile in any up to date machine in the near future... Fedora has been among the first to update the compiler, but the rest will follow shortly.

I can reproduce that issue here now, but doing `export CFLAGS=-Wno-int-conversion` before compiling does I think match the previous behaviour.

If it becomes a bigger problem, I'll revisit of course.

jorgegv, 2024-04-30:
Ah, I see. So you'll just document somewhere that if you have this problem when compiling 2.2, you can just add your workaround and retry. Easier than putting out a new release.

This does not solve the bug in that line (fields are incorrectly initialized in that structure from that field to the end), but I guess if no one has noticed until now, I guess not many people are compiling ZXN code with 2.2 any more...

Thanks for the explanations. I added a note with your workaround on the installation page, see: https://github.com/z88dk/z88dk/wiki/installation#special-note-on-22