z88dk/z88dk #2527: [question] How is z88dk making it possible to parallely build several targets sharing some code?

Fabrizio-Caruso, 2024-05-08:
I am using z88dk (and other compilers) to build in parallel several targets, which share some common source files.

How is z88dk avoiding race conditions on the temporary .o files for the shared files?
Are the .o generated in separate temporary directories? Is it using some more sophisticated mechanism?

Among the compilers I use in Cross-Lib, only z88dk and gcc (native compiler) make parallel builds possible.
With cc65 I get race conditions on the temporary .o files.

suborb, 2024-05-08:
The need to speed up library builds drove a lot of it. I'm sure there's still race conditions about the place, but it mostly seems to work as you've found out as long as dependencies are tracked.

I think your compilation lines are generally of the form: `zcc +zx -o zx.bin  main.c blah.c common.c` so compile and link together - you don't care about the .o files.

In this case, the only thing written locally will be zx.bin - all the object and other intermediate files are kept in /tmp away from any potential collision.

As soon as you start wanting to do incremental builds you need a different approach - the multitarget makefile we have has a subdirectory per target and all the .o files for a target get copied into that - again no clashing.

Fabrizio-Caruso, 2024-05-08:
So you don't even generate .o files when you compile and link with one command. The .o are sort of just generated in RAM memory. Isn't it?

For Cross-Lib my problem is more about the number of different targets and less the single compilation unit. I have to compile sometimes 8 games, each for 100 or even 200 targets. Each single game on a single target is compiled fast enough.

I am asking the CC65 devs to implement support for parallel build.

closed 2024-05-10 by Fabrizio-Caruso

Fabrizio-Caruso, 2024-05-14:
@suborb, how do you avoid collissions in /tmp ? I mean if the files have the same names they would still collide.
Do you generate unique file names? How?