z88dk/z88dk #2528: [bug?] Versions from the same date but Z88DK Linux fails with weird error while Windows version is OK

Fabrizio-Caruso, 2024-05-08:
When I try to build Horde for any Z88DK target with Cross-Lib under Windows/Cygwin  (latest snapshot v22406-6d07223b54-20240508) I have no issue; whereas under Linux, (latest snapshot: v22408-d25a57a490-20240508 through snap) I get this error:
```
games/horde/main.c:2470:31: fatal error: Function `sleep_and_wait_for_input` was already defined at "games/horde/main.c":356
Compilation aborted
z88dk-sccz80: No such file or directory
gmake: *** [Makefile_common:1769: spectrum_48k] Error 1
```
I get the same error under Linux if I build from source. Of course I don't get any problem with `gcc` or any of the other compilers I use.

Under Cross-Lib, you can reproduce this by running: `xl horde spectrum`.

The `sleep_and_wait_for_input` function is only defined once at line 356 and there is nothing fishy in line 2470:
[image]

[image]

I use `sleep_and_wait_for_input` inside several macros. Maybe something goes wrong with the preprocessor? This is a wild guess.
Is the Windows version of Z88DK using a different macro preprocessor? 

Full make command with error here:
```
gmake spectrum USE_TOOLS=1  ZSDCC_MAKEFILE_THREADS_OPTS='-j 5'  ZSDCC_MAKEFILE_COMPILATION_OPTS='-SO3 ' Z88DK_MAKEFILE_COMPILATION_OPTS='-O3' GCC4TI99_MAKEFILE_COMPILATION_OPTS='-O2' NATIVE_MAKEFILE_COMPILATION_OPTS='-Wpedantic -Wall' CMOC_MAKEFILE_COMPILATION_OPTS='-O2' CC65_MAKEFILE_COMPILATION_OPTS='-Or -Cl' LCC1802_MAKEFILE_COMPILATION_OPTS='"-Wf-volatile" -O "-Wp-D nofloats" "-Wa-D LCCNOLONG" "-Wf-mulcall"' _NATIVE_CC=gcc TOOL_CC=gcc GNU_MAKE=gmake -f games/horde/Makefile.horde
games/horde/makefiles/Makefile.override:3: warning: overriding recipe for target 'c16_16k'
Makefile_common:383: warning: ignoring old recipe for target 'c16_16k'
games/horde/makefiles/Makefile.override:24: warning: overriding recipe for target 'creativision'
Makefile_common:554: warning: ignoring old recipe for target 'creativision'
zcc +zx     -O3 -Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/cc65/gamate -Icross_lib/sound/cc65/creativision -Icross_lib/sound/cc65/atari_lynx -Icross_lib/sound/cc65/pet -Icross_lib/sound/cc65/supervision -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/gb -Icross_lib/sound/z88dk/psg -Icross_lib/sound/lcc1802/comx -Icross_lib/sound/cmoc/mo5 -Icross_lib/sound/generic -Icross_lib/sound/gcc4ti99 -Icross_lib/text -Icross_lib/input -Icross_lib/rand -Icross_lib/display/redefine_characters -Icross_lib/display/tiles -Igames/horde/generated_assets -Igames/horde \
-pragma-include:cross_lib/cfg//z88dk/zpragma_clib.inc \
-pragma-redirect:getk=getk_inkey \
-D_XL_SLOW_DOWN_FACTOR=10*160 \
-D__SPECTRUM__ \
-lndos -create-app \
-D__CONIO_GRAPHICS \
-pragma-redirect:fputc_cons=fputc_cons_generic \
-o ../build/Xhorde_spectrum_48k.prg \
cross_lib/display/init_graphics/z88dk/udg/udg_init_graphics.c \
games/horde/generated_assets/z88dk_xchase.asm \
cross_lib/display/display_macros.c cross_lib/input/input_macros.c cross_lib/sleep/sleep_macros.c games/horde/main.c 
games/horde/main.c:2470:31: fatal error: Function 'sleep_and_wait_for_input' was already defined at "games/horde/main.c":356
Compilation aborted
z88dk-sccz80: No such file or directory
gmake: *** [Makefile_common:1769: spectrum_48k] Error 1
```

Fabrizio-Caruso, 2024-05-09:
I am closing this issue because it is a duplicate of #2427.

I have now fixed this error by running dos2unix. 

I think we should somehow enforce unix format for people who use Windows maybe through git.

closed 2024-05-09 by Fabrizio-Caruso

pauloscustodio, 2024-05-09:
The file ```.gitattributes``` lists all file extensions and the corresponding type (binary, text eol=lf, text eol=crlf); maybe we just need to add missing extensions.

Fabrizio-Caruso, 2024-05-09:
@pauloscustodio  how is `.gitattributes` in Z88DK going to fix the macro bug in a different project?
The problem seems to be that the current macro preprocessor is broken in some cases with Windows text files. So probably due to different new lines conventions. So how is the git config going to fix the preprocessor?

pauloscustodio, 2024-05-09:
> @pauloscustodio how is .gitattributes in Z88DK going to fix the macro bug in a different project?
It will not. What I meant is that .gitattributes ensures that no z88dk file has the wrong line feed inside the z88dk tree.

Fabrizio-Caruso, 2024-05-09:
@pauloscustodio what about the macro preprocessor? Couldn't the preprocessor pre-apply dos2unix before evaluating the source files and so fix these bugs?

pauloscustodio, 2024-05-17:
I think the problem was solved by @suborb, by ignoring CR characters.