z88dk/z88dk #2529: [bug][zx81][wrx] When compiling for zx81 with wrx I now get `undefined symbol: hrg_phase`

Fabrizio-Caruso, 2024-05-09:
When I try to build any of my Cross-Lib for the zx81 target with wrx I now get these errors

```
C:/z88dk/bin/../lib/target/zx81/classic/zx81_hrg.asm:20: error: undefined symbol: hrg_phase
C:/z88dk/bin/../lib/target/zx81/classic/zx81_hrg.asm:21: error: undefined symbol: _hrg_phase
make: *** [Makefile_common:1916: zx81_32k_wrx] Error 1
```

This can be reproduced with Cross-Lib by running `xl bomber zx81_wrx`

More details here:
```
make zx81_wrx USE_TOOLS=1  ZSDCC_MAKEFILE_THREADS_OPTS='-j 24'  ZSDCC_MAKEFILE_COMPILATION_OPTS='-SO3 ' Z88DK_MAKEFILE_COMPILATION_OPTS='-O3' GCC4TI99_MAKEFILE_COMPILATION_OPTS='-O2' NATIVE_MAKEFILE_COMPILATION_OPTS='-Wpedantic -Wall' CMOC_MAKEFILE_COMPILATION_OPTS='-O2' CC65_MAKEFILE_COMPILATION_OPTS='-Or -Cl' LCC1802_MAKEFILE_COMPILATION_OPTS='"-Wf-volatile" -O "-Wp-D nofloats" "-Wa-D LCCNOLONG" "-Wf-mulcall"' _NATIVE_CC=gcc TOOL_CC=gcc GNU_MAKE=make -f games/bomber/Makefile.bomber
zcc.exe +zx81     -O3 -subtype=wrx -clib=wrx \
-Icross_lib -Icross_lib/sleep -Icross_lib/display -Icross_lib/display/graphics_mode -Icross_lib/display/alt_print -Icross_lib/include -Icross_lib/sound -Icross_lib/sound/cc65/atmos -Icross_lib/sound/cc65/c264 -Icross_lib/sound/cc65/pokey -Icross_lib/sound/cc65/sid -Icross_lib/sound/cc65/vic20 -Icross_lib/sound/cc65/gamate -Icross_lib/sound/cc65/creativision -Icross_lib/sound/cc65/atari_lynx -Icross_lib/sound/cc65/pet -Icross_lib/sound/cc65/supervision -Icross_lib/sound/z88dk/bit_bang -Icross_lib/sound/z88dk/gb -Icross_lib/sound/z88dk/psg -Icross_lib/sound/lcc1802/comx -Icross_lib/sound/cmoc/mo5 -Icross_lib/sound/generic -Icross_lib/sound/gcc4ti99 -Icross_lib/text -Icross_lib/input -Icross_lib/rand -Icross_lib/display/redefine_characters -Icross_lib/display/tiles -Igames/bomber/generated_assets -Igames/bomber \
-pragma-include:cross_lib/cfg//z88dk/zpragma_clib_no_printf.inc \
-pragma-define:hrgpage=36096  \
-D__NO_SPRITE_TRANSPARENCY \
-D_XL_SLOW_DOWN_FACTOR=10*1 -DNO_ANIMATION \
-DXSize=32 \
-DYSize=22 \
-vn  \
-D__ZX81__ \
-D__Z88DK_SPRITES_GRAPHICS -D__SPRITE_X_STEP=8 -D__SPRITE_Y_STEP=8 \
-lndos \
-D_XL_NO_COLOR \
-D__NO_SCREEN_COLOR_INIT -D_XL_NO_JOYSTICK \
-D__ALT_PRINT \
-D__FEWER_SPRITES \
-create-app -o  ../build/Xbomber_zx81_32k_wrx.prg \
cross_lib/display/display_macros.c cross_lib/input/input_macros.c cross_lib/sleep/sleep_macros.c games/bomber/main.c  \
cross_lib/display/init_graphics/z88dk/zx81/zx81_init_graphics.c \
cross_lib/display/alt_print/memory_mapped_alt_print.c
C:/z88dk/bin/../lib/target/zx81/classic/zx81_hrg.asm:20: error: undefined symbol: hrg_phase
C:/z88dk/bin/../lib/target/zx81/classic/zx81_hrg.asm:21: error: undefined symbol: _hrg_phase
make: *** [Makefile_common:1916: zx81_32k_wrx] Error 1
```

suborb, 2024-05-10:
@zx70 is doing some fairly major tweaks on the zx81 video at the moment, so this is probably a fallout from that.

zx70, 2024-05-13:
Right.  Hopefully the latest snapshot altready includes a fix for it, please let me know.

Fabrizio-Caruso, 2024-05-13:
Hi @zx70, I could get one of my games to work. Have you improved the performance of the WRX driver?
It looks still very slow.
It would be nice to have some sort of UDG (solid, non-trasparent, 8 pixel-aligned tiles). It would make my games playable.
I currently use sprites, which cannot assume they are alligned on 8 pixel boundaries and I have to delete and draw them each single time I draw a "tile".

zx70, 2024-05-13:
there's little space for speed improvements on the driver itself.   I can say that the 64 rows mode is faster and that hrg_blank()  will make the programs run fasted  when the display is blanked.
On the libraries you could achieve a little bit of boost by using clga() to delete the sprites area,  iirc I tuned it to help with the putsprite() slowness.
I also think that the text driver allows the font redefinition in graphics mode,  I don't remember if the udg is supported.

Fabrizio-Caruso, 2024-05-13:
I am not aware of UDG for the ZX81. That is what I need. Hopefully they are faster than sprites.

Is the arx driver meant for a more exotic hardware?

How can I use clga()?

zx70, 2024-05-14:
I was going to integrate my answer, the zx81 had several video modifications to extend it's capabilities. 
The first ones simply aimed to replace the ROM font.
You can look at examples/zx81/dstar.c, I've managed to support most of the possible font positions by pressing the numeric keys: different DIY projects existed on the magazines as well as some commercial interface.
At its maximum extent this mod allows a software trick, the ARX driver (by Andy Rea, recently improved by Fruitcake) which swaps the fonts during the video raster pass to get a full graphics mode.  This means that the ARX graphics library is not your best choice when you have al the font redefinition option in text mode. 

The WRX mode is different, it uses ram areas in different positions which couldn't be used in text mode.  Other historical graphics extensions were the Memotech HRG, the G007 and few other rare and expensive interfaces.

In graphics mode you can speed up the sprites by reducing their size (see microman), running in 64 rows mode (which also requires less ram memory), and depending on the library quality, using clga(), to clean the video portion.

void clga(int tlx, int tly, int width, int height)

Putsprite() runs a little faster when the sprite content
Includes few pixels.

zx70, 2024-05-14:
I noticed that the optimized clga() could be extended to some more target with 'narrow' HRG graphics.
I just did it for the PX4 and the MC-1000 but testing it is a lenghty process.

You can test the code on your favourite target yourself, you need to spot the right 'lst' file (something like 'libsrc/target/mc1000/mc1000.lst'), alter it and rebuild the target library.
The list file must have a line referring to the 'narrow' graphics list:

    @gfx/narrow/gfx_narrow.lst

alter it by inserting the reference to the alternative code BEFORE the gfx list inclusion:

    gfx/narrow/obj/${TARGET}/clrarea2
    @gfx/narrow/gfx_narrow.lst

Then, move into {z88dk}/libsrc and use make to rebuild the specific target libraries and move the resulting lib files into {z88dk}/lib/clibs

Do not try it on low resolution targets (like the Mattel Aquarius), it won't work.

Fabrizio-Caruso, 2024-05-28:
@zx70 I see a significant improvement on the ZX81 WRX target.
Not much on the C128 80col mode (which in my case should not use sprites by redefined characters but which are not possible (yet or may ever) possible on the C128 when controlled by the Z80 (???)

zx70, 2024-06-06:
I think the sprites on the ZX81 should be slightly faster now

Note that on the ZX81 several HRG modes are available:
WRX
ARX (probably the UDG is a better choice here, but putsprite is supported).

..and with some memory constraint:
Memotech WRX (not all the graphics patterns are usable)
G007

zx70, 2024-06-07:
+Commodore 128,  this should speed-up clga(), xorclga() and fillb().
The vertical overflow protection is missing, but I think it is a good compromise already

closed 2024-06-09 by zx70