z88dk/z88dk #2532: msys2 clang64 error compiling

markjfisher, 2024-05-10:
Hi, I'm part of the FujiNet project, just trying to compile some code for an Adam target, and in my msys2 clang64 environment I tried a fresh install and compile, but got this error:

```
cc -std=gnu11 -MMD -Wall -ggdb -O3 -I. -Isrc -I../common -Isrc/c -It -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src -I../../ext/UNIXem/include -O2 -c -o ../common/fileutil.o ../common/fileutil.c
../common/fileutil.c:796:36: error: incompatible function pointer types passing 'int (*)(const char *, int)' to parameter of type 'const int (*)(const char *, int)'
      [-Wincompatible-function-pointer-types]
  796 |     int ret = glob(pattern, flags, errfunc, pglob);
      |                                    ^~~~~~~
../../ext/UNIXem/include/unixem/glob.h:157:21: note: passing argument to parameter 'errfunc' here
  157 | ,   const int     (*errfunc)(char const*, int)
      |                     ^
1 error generated.
```

I wondered if it needed to be a POSIX (i.e. just msys2) shell, but that died immediately:

```
make -C src/z80asm PREFIX=`pwd` PREFIX_SHARE=`pwd` install
make[1]: Entering directory '/d/dev/external/z88dk/src/z80asm'
cc -std=gnu11 -MMD -Wall -ggdb -O3 -I. -Isrc -I../common -Isrc/c -It -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src -I../../ext/UNIXem/include -O2 -c -o src/c/alloc.o src/c/alloc.c
In file included from ../common/fileutil.h:26,
                 from ../common/die.h:8,
                 from src/c/dbg.h:11,
                 from src/c/alloc.h:18,
                 from src/c/alloc.c:12:
../../ext/UNIXem/include/glob.h:72:3: error: #error This file is only currently defined for compilation on Win32 systems
   72 | # error This file is only currently defined for compilation on Win32 systems
      |   ^~~~~
../../ext/UNIXem/include/glob.h:115:9: error: unknown type name ‘unixem_glob_t’
  115 | typedef unixem_glob_t glob_t;
      |         ^~~~~~~~~~~~~
In file included from ../common/fileutil.h:27:
../../ext/UNIXem/include/dirent.h:72:3: error: #error This file is only currently defined for compilation on Win32 systems
   72 | # error This file is only currently defined for compilation on Win32 systems
      |   ^~~~~
In file included from ../../ext/UNIXem/include/dirent.h:79:
../../ext/UNIXem/include/unixem/dirent.h:73:3: error: #error This file is only currently defined for compilation on Win32 systems
   73 | # error This file is only currently defined for compilation on Win32 systems
      |   ^~~~~
make[1]: *** [Makefile:87: src/c/alloc.o] Error 1
make[1]: Leaving directory '/d/dev/external/z88dk/src/z80asm'
make: *** [Makefile:142: bin/z88dk-z80asm.exe] Error 2
```

It's just compiled fine though in UCRT64 msys2 environment, and I'm guessing it would in MINGW64 too.

markjfisher, 2024-05-10:
I'll close this, I had read the install guide which said "see linux below" but there was nothing below, so I bludgeoned on guessing that I could just run make.
Found the instructions for unix and have switched to my WSL2 installation, and the build.sh script seems to be working fine so far.

I'll revisit the msys afterwards to see if I can get that working too.

closed 2024-05-10 by markjfisher

suborb, 2024-05-10:
The installation page was re-arranged in order of ease, so I wonder if some references needed to be updated.

CI is building on msys2 - see https://github.com/z88dk/z88dk/blob/master/.github/workflows/build-on-windows-msys2.yml for the recipe

markjfisher, 2024-05-10:
Thanks, the WSL2 build worked fine, but I like to use msys2 for most things, so this will help.