z88dk/z88dk #2538: MacOS - zsdcc binary is r14210

feilipu, 2024-05-15:
Just playing around with MacOS, and I note that it seems that the zsdcc binary bundled with the daily seems to be r14210.

This extracted from a test compile.
`zcc +cpm -compiler=sdcc samegame.c -o samegame --list -create-app`

``` asm
samegame.c:
                                
     1                          ;--------------------------------------------------------
     2                          ; File Created by SDCC : free open source ISO C Compiler
     3                          ; Version 4.3.0 #14210 (Mac OS X x86_64)
     4                          ;--------------------------------------------------------
     5                          ; Processed by Z88DK
     6                          ;--------------------------------------------------------
     7                          
```

suborb, 2024-05-20:
Looks like I did the tarball, but didn't build the binaries - Windows ones are on the same version so at least I'm consistently bad. 

https://github.com/z88dk/z88dk/pull/2485#issuecomment-1917900042

suborb, 2024-05-20:
I've rebuilt the binaries, so should be good from the next nightly.

feilipu, 2024-05-21:
Fixed. Thanks.

``` bash
~   $ z88dk-zsdcc -v

ZSDCC IS A MODIFICATION OF SDCC FOR Z88DK
Build: 4.4.0 #14648 (Mac OS X x86_64) May 20 2024

sdcc website:
https://sourceforge.net/projects/sdcc/

patch details:
https://github.com/z88dk/z88dk/blob/master/src/zsdcc/sdcc-z88dk.patch

published under GNU General Public License (GPL)
~   $ 
```

closed 2024-05-21 by feilipu