z88dk/z88dk #2539: appmake +zx colliding with binaries of previous builds

zx70, 2024-05-15:
a_BSS.bin   
a_DATA.bin  

Error: Section  overlaps section DATA by 7317 bytes
Error: Section DATA overlaps section BSS by 677 bytes
zx: Aborting... one or more binaries overlap
Building application code failed

zx70, 2024-06-24:
@pauloscustodio I wonder if this could be addressed by Appmake ?
Is it too risky to begin by wiping <progname>_*.bin on the output folder ?

pauloscustodio, 2024-06-24:
> Is it too risky to begin by wiping _*.bin on the output folder ?

Yes, I think so. We don't know what the user might have in the folder.

What is the test case that triggers that error?

zx70, 2024-06-24:
If you build a program needing sections it will create more than a single file (a.bin, a_bss.bin, etc..).
If a different program is later built appmake tries to gather also the previous files, I thought it was using the .map, but apparently things go differently. 
I got this behaviour with zcc +zx -create-app..

feilipu, 2026-11-09:
A section binary older than the map is left out. An earlier DATA or BSS file no longer overlaps the new image. Commit 83e6a3651a.

closed 2026-11-09 by feilipu