z88dk/z88dk #2540: TMS9918 graphics improvements

zx70, 2024-05-17:
I don't enjoy very much in programming the VDP but the current library deserves to be improved.
the new clga() implementation is now reasonably fast, even if it could be rewritten from scratch and be way more efficient.
Good test cases are xample.c and invazion.c (the latter is in the z88dk-ext repo).

To reuse the VDP POKE logic I've extracted it from stencil_render putting it in __tms9918_store.asm.
While I was on it I've inserted some of those delays probably required by the real video chip, (even though I'm not sure it's totally right) and made it colour ready.
I'm noticing some of the examples crashing on the MSX while they run correctly on the SVI-328,  I'm investigating on it, hints are welcome.   

I haven't tested the results on the MyVision and the Casio PV-2000 ATM.

zx70, 2024-05-17:
There's one more thing I'm thinking at, to add a check on top of clg().   When the VDP is in text mode (mode<2), we could forcefully enter in mode 2

suborb, 2024-05-17:
I think it already does?

zx70, 2024-05-17:
> I think it already does?

I haven't checked the code yet, but on MSX, cassette bload model, I had to add the code to switch to mode 2 to get a graphics output. IIRC it was the same on the Spectravideo SVI

suborb, 2024-05-17:
So, looking at it, it will switch mode automatically if in a text mode, BUT by default it's not in any mode so doesn't do anything!

zx70, 2024-05-18:
is there a way at compiler level to force a video mode at start-up like on the ts2068?

suborb, 2024-05-18:
No. I’ve not added it to all targets. 

I’ll add it in for +msx and some others this weekend.

closed 2024-05-19 by suborb