z88dk/z88dk #2549: sccz80 - lhs constant comparison

feilipu, 2024-06-02:
I read about the practice of putting the comparison constant on the lhs to avoid potential silent typo errors arising from `=` rather than `==`. I fixed just such an issue in some other code of mine, so it seems like a worthwhile practice.

So I've had a play with this code to see what happens with sccz80.

```c
int variable;
// long variable;

int main (void)
{
    for (variable = 0; variable < 10; ++variable)
    {
        if ( variable == 0 )
            fprintf(stdout, "%i\n",variable);
    }

    for (variable = 0; variable < 10; ++variable)
    {
        if ( 0 == variable )
            fprintf(stdout, "%i\n",variable);
    }

    return 0;
}
```

For non-zero constants there is nothing to discuss. But where the constant is zero, then two questions arise.

### zero constant RHS

In this case an optimisation is done for `int` comparison, but I don't understand why the code generated does a double `jp` and with the `scf`. Wouldn't generating a `jp nz,i_5` be just as effective?

```asm
	ld	hl,(_variable)
	ld	a,h
	or	l
	jp	nz,ASMPC+4
	scf
	jp	nc,i_5	;
	ld	hl,__sgoioblk
	ld	bc,10
	add	hl,bc
	push	hl
	ld	hl,i_1+0
	push	hl
	ld	hl,(_variable)
	push	hl
	ld	a,3
	call	fprintf
	pop	bc
	pop	bc
	pop	bc
.i_5
```

### zero constant LHS

Here there is no optimisation done for `int` comparison, so there's an opportunity to treat this case in the same way as above.

```asm
	ld	hl,(_variable)
	ld	de,0
	call	l_eq
	jp	nc,i_9	;
	ld	hl,__sgoioblk
	ld	bc,10
	add	hl,bc
	push	hl
	ld	hl,i_1+0
	push	hl
	ld	hl,(_variable)
	push	hl
	ld	a,3
	call	fprintf
	pop	bc
	pop	bc
	pop	bc
.i_9

suborb, 2024-06-02:
The weirdness in the LHS case results from needing to support `x = val == 0` as well as regular flow control.

The comparison helper functions return carry=1, hl=1 or carry=0,hl=0 as the result. When doing a comparison, the carry flag is used for the dropout check, when doing the assignment hl is used. There's 

So in the RHS case here, we set the lval result type to `KIND_CARRY` which is then promoted up if necessary with `rl l, ld h,0`.

I did think some of this was cleaned up by the rules, but maybe not all.

The LHS case is because comparisons use `skim()` but I concentrated the `plnge2a()` and `plnge2b()` when sorting out constant behaviour.

feilipu, 2024-06-03:
> RHS case results from needing to support `x = val == 0`
Ah. Yes, of course it has into fit the standard comparison helper result.

> I did think some of this was cleaned up by the rules, but maybe not all.
I didn't see any asm changes from `-O0` to `-O2`, so perhaps something could be done?

> LHS case is because comparisons use `skim()`
I was reading the `plnge2a()` code trying to see where the different handling originated (couldn't see anything), but I didn't see `skim()` so I'll go back and read further.

feilipu, 2024-06-03:
Based on the following tests, I think the rule for z80rules.0 could to have this added.
```asm
	ld	hl,(%1)
	ld	a,h
	or	l
	jr	nz,ASMPC+3
	scf
	jp	nc,%2	;
=
	ld	hl,(%1)
	ld	a,h
	or	l
	jp	nz,%2	;
```

and for 8085_rules.1

```asm
	ld	hl,(%1)
	ld	a,h
	or	l
	jp	nz,ASMPC+4
	scf
	jp	nc,%2	;
=
	ld	hl,(%1)
	ld	a,h
	or	l
	jp	nz,%2	;
```
If there's an assignment required, then that is done before the second jump instruction, like below example.
So these rules won't trigger.
Thoughts?

```asm
	ld	hl,(_variable)
	ld	a,h
	or	l
	jp	nz,ASMPC+4
	scf
	ld	hl,0	;const
	ld	a,0	;carry
	rla
	ld	l,a
	ld	(_x),hl
	ld	a,h
	or	l
	jp	z,i_4	;
```
```c
int variable = 10;
int x;

int main (void)
{
//  variable = 10;

    {
        if ( variable == 0 )
            printf("%i",variable);
    }

    {
        if ( x = variable == 0 )
            printf("%i",variable);
    }

    {
        if ( x = ( variable == 0 ) )
            printf("%i",variable);
    }

    {
        if ( x = 0 == variable )
            printf("%i",variable);
    }

    {
        if ( x = ( 0 == variable ) )
            printf("%i",variable);
    }

    return x;
}
```

feilipu, 2024-06-03:
Playing around a bit further there's a rule to fix optimise this...
So let's find it.

```c
int variable;
int x;
int y;
int main (void)
{
    {
        if ( y = ( x == 5 || variable == 5 ) )
            printf("%i",variable);
    }
}
```
This at `-O0`
```asm
	ld	hl,(_x)
	ld	de,5
	and	a
	sbc	hl,de
	scf
	jr	z,ASMPC+3
	ccf
	jp	c,i_5	;
	ld	hl,(_variable)
	ld	de,5
	and	a
	sbc	hl,de
	scf
	jr	z,ASMPC+3
	ccf
	jp	c,i_5	;
	ld	hl,0	;const
	jr	i_6
.i_5
```
becomes this at -O2`
```asm
	ld	hl,(_x)
	ld	de,5
	and	a
	sbc	hl,de
	jp	z,i_5	;
	ld	hl,(_variable)
	ld	de,5
	and	a
	sbc	hl,de
	jp	z,i_5	;
	ld	hl,0	;const
	jr	i_6
.i_5
	ld	hl,1	;const
.i_6
```

feilipu, 2024-06-03:
OK. Half Done.

But, I don't have a solution for the more difficult problem of optimising the lhs zero integer comparison.

suborb, 2024-06-03:
I think this is it:

```
diff --git a/src/sccz80/plunge.c b/src/sccz80/plunge.c
index e4fa4758cb..f9c19409e6 100644
--- a/src/sccz80/plunge.c
+++ b/src/sccz80/plunge.c
@@ -398,7 +398,7 @@ void plnge2a(int (*heir)(LVALUE* lval), LVALUE* lval, LVALUE* lval2, void (*oper

     /* Special case handling for operation by constant */
     if ( constoper != NULL &&
-        ( oper == mult || oper == zor || oper == zand || oper == zxor || lval2->is_const) ) {
+        ( oper == mult || oper == zor || oper == zand || oper == zxor || oper == zeq || oper == zne || lval2->is_const) ) {
         int doconstoper = 0;
         int64_t const_val;
```

My mistake, it does all go through plnge2a. I'm sure we could add a special case to swap the operator (so < becomes >) for LHS.

commit 388062b3 (2024-06-06): Merge pull request #2558 from z88dk/Issue_2549_lhs_const_compare

Fix #2549 - cleanup LHS const != and ==
closed 2024-06-06 by suborb