z88dk/z88dk #2552: ticks - 8085 unimplemented RIM causing regex test failure

feilipu, 2024-06-04:
I've added some specific 8085 [code for `l_eq` and `l_ne`](https://github.com/z88dk/z88dk/commit/582664520602a7e34a8149490e9e53214456afa3) that required a [new `test8085_clib.lib`](https://github.com/z88dk/z88dk/commit/9340f80fe1583dff9de14765a61f128a10ca20d5) to be built, since the interface for these new functions is using BC rather than DE registers taking advantage of `sub hl,bc`.

Now the regex test is failing because z88dk-ticks does not implement the `rim` instruction. No idea why this is happening, as I can't see why a `rim` (0x20) instruction would be being used anywhere in any of the crt0 code.

I'm still looking for the solution, but any help would be welcome.

suborb, 2024-06-04:
Wonder if there's an mis-assembly happening with jr nz?

feilipu, 2024-06-04:
Its a long way round (a bit of jenga going on) to get the test building properly with 8085 throughout.

But I think I've tracked it down to a `push hl` being swallowed by unrelated rules.

feilipu, 2024-06-04:
I checked in the changes to make sure that the libraries are being built consistently with 8085 `l_eq` and `l_ne`.
The regex error is now different. But it is now the code generating the above assembly.

```bash
regexp(3): regexp too bigmake[2]: Leaving directory '/home/runner/work/z88dk/z88dk/test/suites/regex'
```

suborb, 2024-06-04:
> Of interest all of the complete built files have 33 push and 31 pop, though that may just be a red herring.

That's expected if there's long values involved.

closed 2024-06-04 by suborb

feilipu, 2024-06-05:
Thanks. That's a good fix for the issue. Much appreciated.

It does leave me curious though, as to which optimisation path I missed?
I'll probably poke around some more to see if I can find it.

---

**Results**

| Test | Z80 | 8080 | 8085 Yesterday | 8085 Today | Improvement |
| --- | --- | --- | --- | --- | --- |
| string | 1266023 | 1673693 | 1611729 | 1610957 | 0% |
| md5 | 43029351 | 57366180 | 47264193 | 46929649 | 0.7% |
| regex | 1968512 | 2276653 | 2242505 | 1932373 | 13% |

commit fc954af1 (2024-06-04): Merge pull request #2555 from z88dk/Issue_2552_8085_opts

Fix #2552: Fix up so that l_eq/l_ne use dehl entry (8085)