z88dk/z88dk #2564: (classic) KC graphics not fully ix independent

zx70, 2024-06-11:
The KC target required the ixiy workaround.
I recently adjusted the w_draw calls which were we bound to the 'narrow' portable versions rather than the 'wide' ones.

We still need to adjust the stencil calls, e.g. w_stencil_add_side is still bound to the w_line->w_line_r.

zx70, 2024-06-11:
I managed to refresh **w_line_r** to use a single index register, apparently it is the only function using IY (-ixiy eventually does the swap).
I'm saving the value in the bss_graphics section, I think it is correct ?

This probably fixes also new the ZX81 library in 384 rows mode.

suborb, 2024-06-11:
Yeah bss_graphics is the right section since it'll be shunted up high as necessary for some targets.

I thought +kc was using the compiled C implementation of the drawing primitives?

zx70, 2024-06-11:
> I thought +kc was using the compiled C implementation of the drawing primitives to avoid the index register issues.
It was doing it in most of the cases, but the stencil part calls w_line_* directly and skips the portable primitives.

For what I could see we had only one file in the 'wide' folder dealing with IY, I thought it was worth to changing it.

suborb, 2024-06-11:
> It was doing it in most of the cases, but the stencil part calls w_line_* directly and skips the portable primitives.

Ah.....I missed that path.

zx70, 2024-06-12:
The new wide-rez optimized code fits perfectly on the Microbee  :)
I know I'm ignoring the color attributes topic but they'd be too many changes at once for me.

zx70, 2024-06-12:
The new clga() etc optimized code runs very well on the Microbee.
Some of the existing code can be probably replaced with the new one, it contains unfinished opts:
https://github.com/zx70/z88dk/blob/624fe9ca6d86dae135e08b903d99981434da1127/libsrc/target/bee/graphics/w_stencil_render.asm#L207

(w_pixeladdress is what the memory banking mess just above is trying to avoid).

zx70, 2024-06-12:
@suborb , I have a question regarding the colour attributes.  If Inunderstood correctly they are inherited by the graphics from the same value of foreground+background used in the console text,  right?
supposing it affect plot() it should be applied to all the graphics functions, like putsprite() is it correct?

suborb, 2024-06-12:
Yes, that's right, it was easiest to take the attribute value from there.

It should affect all graphics functions yes.

closed 2024-06-16 by zx70